FreeCAD/FreeCAD #15660: Stylesheet: Bounding box not visible with Light theme

maxwxyz, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using FreeCAD Light the bounding box in e.g. check geometry is a white dotted line on white background and not visible:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

knipknap, 2024-07-28:
And something that may or may not be related: I cannot see text selections in any text entry fields either using the same theme.

(Light Theme, Overlay style sheet: Auto)

Non-highlighted text: Black + white background
Highlighted text: Black + white background

When selecting text in the log window:

Non-highlighted: Red + white background
Highlighted: Black + white background

Could be a separate bug if the root cause is not the same.

maxwxyz, 2024-07-28:
@knipknap it's fixed in the lates weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

knipknap, 2024-07-28:
Oh, great! Thanks for the heads-up. Didn't use that one because of #15538, but looking forward to an upgrade ;-).

MisterMakerNL, 2024-07-28:
Seems to be noted here https://github.com/FreeCAD/FreeCAD/blob/00990cb1e193bb5443f8288d7932d5a266f1ec21/src/Mod/Part/Gui/DlgSettingsObjectColor.ui#L396

FEA-eng, 2024-08-03:
This got fixed by #15586, I tested it in the newest Windows weekly.

closed 2024-08-03 by luzpaz