FreeCAD/FreeCAD #15662: Sketcher WB can't delete external geometry

mfraser, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Sketcher WB if you use the "create external geometry" tool to link to other geometry and then try to delete it the element isn't deleted.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38340 (Git)
Build type: Release
Branch: main
Hash: 58b3d0217f92c4dbf2e1bc4c42a31afcf4564c24
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.2.7
  * Gridfinity 0.5.0
  * fasteners 0.5.25
  * dodo 1.0.1
  * Curves 0.6.42
  * ThreadProfile 1.90.0
  * sheetmetal 0.4.23
  * VendorParts
  * lattice2 1.0.0
  * Freecad-Built-in-themes-beta 1.2.2
  * Lithophane
  * ProDarkThemePreferencePack 1.0.0
  * Manipulator 1.5.7
  * DynamicData 2.62.0
  * FreeCAD-themes 2024.7.24
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
  * Assembly4 0.12.4
  * freecad.gears 1.2.0
  * A2plus 0.4.65
  * fcgear 1.0.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

In the attached file. If you open Sketch001 and try to delete ExternalEdge1 it will not be deleted.
[External Test.FCStd](https://github.com/user-attachments/files/16403052/External.Test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

knipknap, 2024-07-28:
Cannot reproduce in 0.22.0dev.37898 (Git) AppImage.
Also cannot reproduce in 0.22.0dev.38334 (Git) AppImage.

mfraser, 2024-07-28:
> Cannot reproduce in 0.22.0dev.37898 (Git) AppImage. Also cannot reproduce in 0.22.0dev.38334 (Git) AppImage.

It seems to work OK with the FreeCAD_weekly-builds-38334-conda-Linux-x86_64-py311.AppImage so it could be a build issue at [the FreeCAD daily PPA](https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily)

nazar-pc, 2024-07-28:
I believe I saw issues like this in https://github.com/FreeCAD/FreeCAD/issues/15042 as well, which was recently fixed, maybe PPA has an older version prior to this fix.

maxwxyz, 2024-07-28:
Cannot reproduce either and should be fixed. Please try in the latest weekly.

maxwxyz, 2024-07-29:
please try with tomorrows weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

mfraser, 2024-07-29:
> please try with tomorrows weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

It is already working in version: 0.22.0dev.38334 (Git) AppImage from the weekly-builds.

maxwxyz, 2024-07-30:
it's now present for me in 
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.23

```

Can anyone confirm with the latest weekly that it is back again?

mfraser, 2024-07-30:
It works in FreeCAD_weekly-builds-38334-conda-Linux-x86_64-py311.AppImage released yesterday.

nazar-pc, 2024-07-30:
38334 was released on July 25th, not yesterday (30th)

mfraser, 2024-07-30:
> 38334 was released on July 25th, not yesterday (30th)

If I go to the weekly builds, it says that was released yesterday.

maxwxyz, 2024-07-31:
@PaddleStroke I've reopened and assigned you. The latest weekly already has the issue that you cannot delete external geometry in a sketch.

mfraser, 2024-07-31:
> 38334 was released on July 25th, not yesterday (30th)
The x86_64 build for 38375 didn't build for some reason.

closed 2024-08-02 by chennes