FreeCAD/FreeCAD #15663: Can't create an external geometry from a linked sketch belonging to another document

nazar-pc, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Setup is simple:
* first file with a sketch
* second file with a link to above sketch
* sketch in second file tries to create external geometry from a linked sketch, but fails

```
16:51:20  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'TypeError'>: Datum feature type is not yet supported as external geometry for a sketch
16:51:20  Test#Sketch: App.getDocument('Test').getObject('Sketch').addExternal("Link","Edge4")
16:51:20  Test#Sketch: Failed to add external geometry
```

Demo:

https://github.com/user-attachments/assets/acd862c9-e317-4239-a961-89fa758e8ba0

Files used in demo:
[LinkedSketchExternalGeometry.zip](https://github.com/user-attachments/files/16403945/LinkedSketchExternalGeometry.zip)

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

Sketcher

### Anything else?

The only mention of this that I found is in https://forum.freecad.org/viewtopic.php?style=10&t=72596, but I'm not in PartDesign, these are simple sketches at the root of the document.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
Was this possible in the stable 0.21.2 version?
@PaddleStroke FYI, do you know if external geometry from a linked sketch in a different document is possible?

nazar-pc, 2024-07-28:
Stable doesn't run with latest Mesa on Linux, so can't test.

Workaround I found was to create sub-object(s) shape binder, though it would have been so much better to not do that.

FEA-eng, 2025-07-17:
I can't even select the geometry of that copied sketch.

FEA-eng, 2025-09-17:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

Plasmarobo, 2026-01-04:
I've also run into this. Can't use linked objects as external geometry, event from within the same part.

PaddleStroke, 2026-01-21:
While subshapebinders can do the trick, IMO it should work without them. There is no reason to require them if it can work directly on Links.

As for bug vs feature, I think both analysis are correct. It can be considered a bug and it can be considered a feature request. IMO probably more a feature request, even if the error message is cryptic. But if the error message was clear like "External geometry does not support Link yet" then it would be clear that it's a feature request.

Anyway the distinction bug / feature request doesn't really matter. The only question is, do we merge it for 1.1 or not. And for this the real question is : is it risky to get in 1.1. Since it currently doesn't work at all in 1.1, and since it does not touch existing behavior then worse case it doesn't work in some cases, but as it's all new then I guess it's ok.

Besides, this functionality is something that is asked for assembly for top-down design. See https://github.com/FreeCAD/FreeCAD/issues/17951
However I wonder @theosib how well it works in the case of assembly? Because in that case both objects are links : The reference geometry, and the sketch. So is the placement of external correct in this case?

theosib, 2026-01-21:
If we require subshape binders for links and array elements, why don't we require them for ALL geometry?

I think the main use of subshape binders right now is to be able to extrude faces from solids. By that I mean select a face, and extrude. Even then, I think subshape binder is a hack, since I'd expect that any flat face is a valid 2D geometry that should be extrudable.

FEA-eng, 2026-09-30:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```