FreeCAD/FreeCAD #15665: OpenGL problem - The new CAM-Simulator don't work an Linux-System with Wayland

michaelnum, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

it seem that something in den OpenGL implementation is wrong, because the New CAM-Simulator does not work on systems with wayland. I does not move, stand only on the first picture. if you minimize/maximize an other application it works in the background picture for picture (with low CPU load).
If you start FreeCAD with setting MESA_LOADER_DRIVER_OVERRIDE it is working, but not smooth with 100% CPU-Load.

That depends in the newest 0.22 builds.
It affects Appimage and the Snap-Edge Builds
 
First i see this an a Debian 12.6-system. Than i tested Kubuntu 20.04, 22.04 and 24.04 and Xubuntu 24.04. With Intel and AMD graphics:
It works correct only on Kubuntu 20.04, that have the X11-System.
All other have wayland and the same problem. On Xubuntu is the same, the means is is not a problem from KDE

Is there an workaround?

In the forum [Known OS specific Issues](https://forum.freecad.org/viewtopic.php?t=30573) it looked, but the  nothing similar help.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-28:
@shaise FYI

shaise, 2024-07-28:
Thanks @maxwxyz ,
I saw it. I'm still waiting for the maintainers decision regarding my latest PR. It solves quite a bunch of issues, it might solve this as well.

michaelnum, 2024-07-29:
Yes, please integrate it in V1.0 .

Now i build you build the CamSimulator3 branch from the repository from shaise: **Yes it solves the problem**.

shaise, 2024-07-29:
Great!
Another reason for the maintainers to merge it.

mkproteus, 2024-07-31:
I also did a test on opensuse leap 15.6 and have the same behaviour. Both versions 0.22.0dev and CamSimulator3 behave the same. A small debug session with eclipse IDE showed that I get no values in the keyboard variables. There is still a case statement where some values are evaluated, there the variables are without content.
In my system is an AMD graphics card which is addressed with the driver Mesa and it runs X11, no Wayland.
I have tried it on different computers. All computers with Nvidia cards, under windows and Linux leap 15.6 work. AMD with Mesa and X11 also exhibit this behaviour.
Thanks for the great work on this simulator.

shaise, 2024-08-01:
> I get no values in the keyboard variables. There is still a case statement where some values are evaluated, there the variables are without content.

can you explain? I do not understand what the 'keyboard values' mean.

michaelnum, 2024-08-24:
Now, after merging the update form the simulator, i tested the newest appimage build 38506.

But it don't work for me with Debian 12.6 on two systems with Intel and AMD graphics, too.
Direct complied from your branch https://github.com/shaise/FreeCAD/tree/CAM-Simulator3 it's working (i synchronized it today).

It seems there is another problem.

JohnOCFII, 2024-08-24:
Testing with a self-built build of FreeCAD on MacOS, it seems some older behavior is back.  The model isn't centered in the window, and I can't see a way to zoom in or out.  Also -- it would be nice if you could hover the mouse over the icons to see what their function is.
[image]

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38507 (Git)
Build type: Release
Branch: main
Hash: baec1037d8ce9c02f6dc67685926fce9dbaf42cf
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.8.17
```

shaise, 2024-08-24:
Is there anywhere an updated tutorial how to compile on ubuntu from scratch? I use the official [freecad](https://wiki.freecad.org/Compile_on_Linux) one, but it just causes lots of issues.

luzpaz, 2024-08-24:
@shaise notice that the @JohnOCFII's report is for macOS.  
Regarding Ubuntu, perhaps @kkremitzki may know ?

shaise, 2024-08-24:
> Testing with a self-built build of FreeCAD on MacOS, it seems some older behavior is back. 
@JohnOCFII can you send me the problematic file? perhaps the centering problem is not only on mac.

shaise, 2024-08-24:
@michaelnum , it works for me on ubuntu 22 + nvidia. I will try to find the differences.

shaise, 2024-08-24:
@michaelnum , it works for me on ubuntu 22 + nvidia. I will try to find the differences.
[image]

JohnOCFII, 2024-08-24:
> @JohnOCFII can you send me the problematic file? perhaps the centering problem is not only on Mac.

Also - here's what I see with the older simulator.
[image]

FreeCAD file is attached.
[Cam Test 2.1.FCStd.zip](https://github.com/user-attachments/files/16738043/Cam.Test.2.1.FCStd.zip)

JohnOCFII, 2024-08-25:
> Do you have any control on the view? Panning (middle drag), zooming (scroll wheel), rorating (middle-left drag)?

I don't have a 3-button mouse.  (I usually don't use any mouse at all, and just use the MacBook Pro trackpad - I use the TouchPad settings within FreeCAD).  I attached a USB 2-button mouse, but behavior is the same as I see with the trackpad.  If I use the scroll wheel to zoom, It does zoom in and out, but as I can't move the object, when I zoom out to the point where I should be able to see the object, it is outside of the viewable area, as I can't seem to pan without a middle mouse button.

In FreeCAD, using the TouchPad settings, I pan by holding the SHIFT key and moving the mouse (or finger on trackpad).  I rotate by holding the OPTION key and moving the mouse (or finger on trackpad).

> Edit: the attached file has no Job in it.

That's odd!  I see the job, as shown in the screen shot above.  I've re-saved it just in case saving with the newer version makes it appear differently.

[AugustToronto_JewelBox.FCStd.zip](https://github.com/user-attachments/files/16741924/AugustToronto_JewelBox.FCStd.zip)

shaise, 2024-08-26:
@JohnOCFII if you have a 2 button mouse with wheel, try clicking the wheel to drag. (just want to make sure it works on a mac as well)
Regardless of that I will add touchpad support as well. This is important. If you can, please open a different issue for that.

JohnOCFII, 2024-08-26:
> try clicking the wheel to drag. (just want to make sure it works on a mac as well)
> Regardless of that I will add touchpad support as well. This is important. If you can, please open a different issue for that.

Interesting!  If I attach the mouse, and press down (click) the wheel alone, I can't drag, but if I then ALSO use the SHIFT key on the keyboard (which is used for PANNING, with the FreeCAD TouchPAD setting) I **CAN** Pan!  I can't zoom out quite as far as I might light, but it gets me a lot closer to a usable system!
[image]

I will open a separate GitHub issue to request TouchPad support.

Thanks!

shaise, 2024-08-26:
> Interesting! If I attach the mouse, and press down (click) the wheel alone, I can't drag

I think for that to work you need to switch from touch pad mode to freecad mode.

shaise, 2024-08-28:
zoom range and direction should be fixed once pr #16118 is merged

shaise, 2024-09-10:
I plan to reduce anti aliasing sampling as a fix for a similar problem. lets see if it will work for Wayland as well.

shaise, 2024-09-17:
@profknivre , Can you check if it works now?

profknivre, 2024-09-17:
It works now. Thank you.

shaise, 2024-09-17:
Great!!

@luzpaz , I think this issue can finally be closed now.

closed 2024-09-17 by luzpaz