FreeCAD/FreeCAD #15667: Visibility issue - Sketch in a face being almost transparent when defining new geometry

lucas-bueno777, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I create a simple extrude in the XY plane and try to sketch in its upper face (aligned with the Z axis), whenever I make new geometry, the preview is barely visible:

Steps to reproduce:

1. Create a rectangle in the XY plane
2. Extrude it along the Z-axis
3. Sketch in the upper face
4. Click to add a new rectangle anywhere in the face
5. The preview of the rectangle is barely visible

In general, the illumination of geometries aligned with the Z-axis leads to many visibility issues.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: Portuguese/Brazil (pt_BR) ]
Installed mods: 
  * A2plus 0.4.60k
  * Curves 0.6.23
  * ExtremeProDark 2.5.3
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

knipknap, 2024-07-28:
I think it matters what theme and override stylesheet you use. Try to include that from the settings.

maxwxyz, 2024-07-28:
@MisterMakerNL @PaddleStroke didn't this issue came up and was fixed? Seems to happen in classic.

Syres916, 2024-07-28:
This is correct in the core code to 127,127,127 / 2139062271 / 0.5f, 0.5f 0.5f but in both https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/PreferencePacks/FreeCAD%20Classic/FreeCAD%20Classic.cfg#L112 as well as https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/PreferencePackTemplates/Sketcher_Colors.cfg#L25 it's not correct and therefore overwriting the core code when the user clicks on FreeCAD Classic.

Should there be any color settings like this stipulated in FreeCAD Classic?

maxwxyz, 2024-07-28:
no, IMO themes should be adjusted.

lucas-bueno777, 2024-07-28:
I forgot to mention, but I selected the classic theme from the start menu.
[image]

Another (maybe unrelated) thing regarding colors is that everything is white when looking from the top view. From other standard views, the default settings lead to unnatural bright scenes as well:

Isometric (reference)
[image]

Top
[image]

Left
[image]

closed 2024-08-12 by WandererFan