FreeCAD/FreeCAD #15668: PartDesign: Pattern order problem

Pendelf, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Pattern functions do not work correctly if you select actions from bottom to top in the operation tree
[image]
If you select from top to bottom, everything works fine
[image]
In the Realthunder (Link Branch) the order of selection does not matter, in any case everything works correctly
[image]

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.5
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

PartDesign

### Anything else?

https://youtu.be/-cJOB5R8hjg

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pendelf, 2024-07-28:
And if you select from bottom to top, the program automatically selects the axis (in this case Z), if you select from top to bottom, the axis is not detected and it must be selected manually, the video shows it

NomAnor, 2024-07-29:
> Pattern functions do not work correctly if you select actions from bottom to top in the operation tree

I don't think this has anything to do with the order in the tree. If I recall correctly, the actions are applied in the order of the list in the pattern dialog ([code](https://github.com/FreeCAD/FreeCAD/blob/04acc05fcbfb7ecfa9dc62fdf5a1bd33180811c4/src/Mod/PartDesign/App/FeatureTransformed.cpp#L327)). In case of

1. Fillet
2. Pocket

First the tool shape of the fillet is patterned, this would effectivly do nothing, because it propably is a boolean union with those corner shapes. Then a boolean cut with the pocket shape is done and the result is just the pockets.

In case of

1. Pocket
2. Fillet

First a boolean cut with the pocket shape is done. And then a boolean union with the fillet shape that will now fill in some part of the pockets.

In Link Branch there is no list in the gui that would define an order, so maybe the order is calculated in the code based on some heuristic, or maybe it uses the order from the tree?

> And if you select from bottom to top, the program automatically selects the axis (in this case Z), if you select from top to bottom, the axis is not detected and it must be selected manually, the video shows it

This has to do with the kind of feature that is selected when the pattern is created. I think only sketch based features provide an initial axis ([code](https://github.com/FreeCAD/FreeCAD/blob/04acc05fcbfb7ecfa9dc62fdf5a1bd33180811c4/src/Mod/PartDesign/Gui/TaskPolarPatternParameters.cpp#L100)). Same for other patterns.

This is probably not a bug, but a discrepancy between the users expectation and what the program actually does.
There might be cases when the order of actions is important and needs to be controlled by the user.
To better show that the ui list defines an order, numbers could be shown before the text, like the ordered lists above. Also a label for the list widget like "Order of features" might help. In Addition, some buttons to change the order would also be helpful.

Pendelf, 2024-07-31:
As usual I tried to do the same in other CAD and there the order of selection does not matter, it seems to me that in Link Branch this function is realized more clearly and conveniently

Pendelf, 2024-12-04:
It's still relevant

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.33
  * PieMenu 1.10.1
  * sheetmetal 0.5.10

tarman3, 2024-12-04:
Confirm. But not sure, this is a bug or feature, that FreeCAD do not automatic sorting selected operations.
[15668_pattern_sort.zip](https://github.com/user-attachments/files/18005958/15668_pattern_sort.zip)

[image]

Pendelf, 2024-12-04:
@tarman3 
If you need the previous operation without the next one, you can select only it (select a pocket without rounding)

Pendelf, 2025-07-13:
Still relevant

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42531 (Git)
Build date: 2025/07/11 16:00:50
Build type: Release
Branch: main
Hash: 759bad7455a3a554cdf48a8182709cad991c8fe5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.63
  * fasteners 0.5.38
  * PieMenu 1.12.0
  * sheetmetal 0.7.23
  * StructureTools 0.0.1
  * SvgWorkbench 1.0.0.dev6
  * Telemetry 1.0.1

FEA-eng, 2025-09-14:
A file would be helpful.

Pendelf, 2025-09-15:
@FEA-eng 

[Polar Problem.zip](https://github.com/user-attachments/files/22327031/Polar.Problem.zip)