FreeCAD/FreeCAD #15670: Can't edit sub-object shape binder support

nazar-pc, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Simply can't edit "Support" property:

https://github.com/user-attachments/assets/8e814ec9-5d6d-4fee-97e2-a6a153d73d3f

I believe it worked when I tried it a couple of weeks ago with whatever was the current dev build back then, so might be a relatively recent regression.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-07-29:
Not a regression.

RTM:
https://wiki.freecad.org/PartDesign_SubShapeBinder#Notes

The current solution is perhaps not very user-friendly, but is intentional. Note that you can also change the ReadOnly attribute of the property in the [Property_editor](https://wiki.freecad.org/Property_editor#Context_menu).

nazar-pc, 2024-07-29:
Thanks, I wasn't aware of that, it felt broken.

It was also not very intuitive, but simply making it not read-only both fixed UI glitch and allowed me to edit it just fine. If editing works, why is it read-only by default?

Roy-043, 2024-07-29:
Forum post with an explanation:
https://forum.freecadweb.org/viewtopic.php?f=3&t=63201&p=543264#p543264

nazar-pc, 2024-07-29:
Thanks, but it makes little sense to me. It is either supported (should not be read only) or it is not (in which case setting to to not read only should be invalid). I don't understand how drag and drop is any different from selecting the features explicitly. Also the big feature of this particular binder is to be able to bind only certain features of elements and I'm not quite sure how I would do that with drag and drop.

Definitely something that is confusing to me as a user and can be improved.