FreeCAD/FreeCAD #15671: Sub-object(s) shape binder doesn't bind points of a sketch in some cases

nazar-pc, 2024-07-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This might be related somehow to the migration prompt I keep getting as described in https://github.com/FreeCAD/FreeCAD/issues/15655, but the gist of it is that in that document sub-object(s) shape binder created on a sketch ignores points, while in a fresh document it doesn't, here is demo:

https://github.com/user-attachments/assets/38f301c6-d497-4e1c-892b-2bb62ef38f06

The problematic file: [SubObjectsPointShapeBinderDemo.zip](https://github.com/user-attachments/files/16405123/SubObjectsPointShapeBinderDemo.zip)
The working file can be created easily on demand.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nazar-pc, 2024-08-04:
I tried to select points explicitly on top of selecting the sketch itself and it still ignores them for some reason :thinking:

Syres916, 2024-08-04:
It's highly likely that the `ToDo` in https://github.com/FreeCAD/FreeCAD/commit/c8046b1c60c786caca3050cf51f853454ffb5388 will be the time when this is then fixed. Basically LinkStage3 doesn't have that code in it but as crashes were being reported left. right and centre it was merged to keep those reports at bay while the LS3 code was still being brought in. This issue has a lot of similarities to https://github.com/FreeCAD/FreeCAD/issues/15735 where I provided a workaround which isn't probably any use in this situation??

nazar-pc, 2024-08-11:
I need to bring multiple points from a sketch as external geometry. Since I can't use liked sketch directly due to https://github.com/FreeCAD/FreeCAD/issues/15663, I resorted to (otherwise unnecessary) shape binder, but now with shape binder I can't have points, so another blocker. I wish either of these two options worked so I can move forward without duplicating things.

If it was one point I could maybe work around it by adding a line to the point temporarily, but I have 22 of them and adding 22 lines to already busy sketch is a hack I'd rather avoid here.

nazar-pc, 2024-12-08:
Still happening on latest weekly:
```
OS: Ubuntu 24.04.1 LTS (Custom/custom/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

FEA-eng, 2026-10-27:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```