FreeCAD/FreeCAD #15673: Lag on Spreadsheets CTRL+A Select All Command

MAJMAD, 2024-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I noticed a lag in the CTRL+A keybind for select all in Spreadsheets. I was playing around with the EasyAlias macro and was testing the binding to see if I could make it a hotkey to improve the macro's use and workflow, and noticed the lag, as it frozen up the cell selections. Initially tested on a small 3x2 cell selection, and since have confirmed it seems to happen anytime a cell is selected. I expect that this might be some of Python's inherent slowness showing while having to touch the entire 16384x702 sheet. I cannot immediately confirm its not just an issue on my machine, perhaps the way the spreadsheet is being handled in memory could be optimized to reduce the lag when executing the command.

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

Thanks for the efforts to make a great tool greater

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-29:
Is this lag due to the macro ?

Roy-043, 2024-07-29:
The problem here is that not only the cell range with content is selected.

luzpaz, 2024-09-13:
> Strictly speaking, freeze does not take place. When you select all cells and apply something, FreeCAD enumerates all the cells in the spreadsheet, taking some time to go through 7.4 GiB of memory (26 * 26 * 16384 * 700 bytes). It is carried out in the GUI thread; therefore, making UI unresponsive. FreeCAD returns back to normal when it is finished.

Per @xtemp09 in https://github.com/FreeCAD/FreeCAD/issues/16494#issuecomment-2348165224

luzpaz, 2024-10-04:
@wwmayer if you have a moment, could you take a look at this ?

luzpaz, 2025-01-13:
This is a usability nightmare. Lets prioritize this.

luzpaz, 2025-01-13:
@mwganson any chance you can look at this ?

luzpaz, 2025-01-14:
@yorikvanhavre can we prioritize this? Users are encountering this very often and are under the wrong impression that FC is crashing.

luzpaz, 2025-01-22:
Getting multiple reports from different sources (social media included) where users are getting bit by this bug. Marking this as a blocker because it disrupts workflow significantly

luzpaz, 2025-01-23:
@tritao any chance you could take a look at this nasty bug? TIA!

hyarion, 2025-02-11:
@luzpaz assign it to me

zayats1, 2025-04-19:
It was probably fixed in 1.1.0 dev 41314

xtemp09, 2025-04-19:
@zayats1, have you compared different versions or just tried to reproduce with this one?

zayats1, 2025-04-19:
@xtemp09 I have tried to reproduce it in this one. Compared to 1.0 , 1.1 is a bit smoother, although I have a pretty decent PC , and the CPU usage spikes were barely noticeable.
However CTRL + A and deletion have freezes in both versions

luzpaz, 2025-05-01:
soft bump

maxwxyz, 2025-08-03:
By definition this is not a blocker. Leaving this label for next discussion.

hyarion, 2025-10-16:
I've unassigned me now to allow others to work on it, thanks for the poke