FreeCAD/FreeCAD #15675: Ruled surface fails offset on any axis other than XY

chopsywa, 2024-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to offset a ruled surface, it will always offset in the XY, regardless of orientation. Therefore, without mucking around with placement parameters manually, you can't use offset on a ruled surface in any plane other than XY. This issue does not occur when offsetting a surface made from wires.

This also affects other operations such as project surface, etc.

I have attached a little file to show the issue. It has three rectangles. First is XY, second is XZ broken, third is XZ with placement values manually modified.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Curves.backup1706353084.2536438 0.6.13
  * fasteners.backup1718262130.4318058 0.5.22
  * OpenDark.backup1718262281.5371034 2024.3.13
```

### Subproject(s) affected?

Part

### Anything else?

[Offset-Error.FCStd.zip](https://github.com/user-attachments/files/16407779/Offset-Error.FCStd.zip)

File is just renamed with zip extension.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-14:
@maxwxyz Might be good to close this one since the issue is much broader than stated here and described in #17235

wwmayer, 2024-10-15:
This is a regression since v0.22.

In v0.21 it worked correctly. If you load the attached project in v0.21 and touch all three sketches then after a recompute the Offset XZ jumps to the correct position. The Offset-Manual-Fix jumps to the xy plane because of the manually set placement.

tomate44, 2024-10-21:
In case it was not noticed, this only happen when the 2 Wires (or Edges) of the Ruled Surface belong to the same object.
In that case, the BSpline surface is built on the XY plane, and an non-valid face is created (with a huge tolerance), since the wire is not lying on the surface.