FreeCAD/FreeCAD #15680: Daily snap builds need to be reenabled

furgo16, 2024-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It appears that the snap daily builds were disabled yesterday due to the repo not being active in the last 60 days. I'm not sure if this is a new GitHub policy, but daily testers would benefit from reenabling the builds.

[image]

Reporting it here, as issues at https://github.com/FreeCAD/FreeCAD-snap/ don't seem to be monitored

### Full version info

```shell
N/A
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-29:
I;'ll take a look

luzpaz, 2024-07-29:
Renabled. Thanks for the heads up

closed 2024-07-29 by luzpaz