FreeCAD/FreeCAD #15681: Draft: for certain Draft_Scale operations a clone must be a compound or nested in a compound

Roy-043, 2024-07-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If a 'randomly' placed object is scaled unequally, and the "Create a clone" checkbox is checked, the Draft_Scale result is inaccurate. There are some bugs in the code, but even if those are fixed, it is obvious that if the clone of the object is not a compound (or nested in a compound or a similar container) the problem cannot be solved.

Test A:
1. Removed the zip extension and open the attached [file](https://github.com/user-attachments/files/16419023/clonetest.FCStd.zip).
2. Select the Sketch.
3. Start the Draft_Scale command.
4. Pick the center of the red circle.
5. In the next task panel check "Create a clone".
6. Enter the scaling factors: X=2, Y=4, Z=6.
7. Notice that the preview matches the blue clone (based on a clone of the Sketch in a Std_Part container).
8. Press OK.
9. Result: the new object is wrongly scaled and misaligned obviously. But whatever scaling or alignment is applied the angles of the edges of the clone will always stay 90 degrees.

Test B:
1. Repeat the test but now select the Sketch and the red circle.
2. Result: because two objects have been selected the clone is a compound and the result of the operation is accurate.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=89439

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marcuspollio, 2024-09-24:
Hi @Roy-043 

Currently, by default, the Scale Tool in Draft and BIM WBs reports an error when used with Solid shape Objects and even some closed Wire shape ones (e.g. circle) because it requires the option "Create a clone" to be checked :

`Unable to scale object: Wall - This object type cannot be scaled directly. Please use the clone method.`

I don't know if, beyond the specific forum issue reported here, it's a record for future modifications to the tool ?
Otherwise, I can open another issue for the above in my message. Let me know what you prefer. Thanks

Roy-043, 2024-09-26:
Thank for mentioning this. We should tackle that as well. It is not very user friendly. No need to create a separate issue.

PR #18795 (2025-01-20): Draft: make move, rotate and scale commands link-aware
Fixes #12836.
Fixes #15681.

These changes are presented as a single ('big') PR because the mentioned commands have several similarities. Working on them seperately would have made less sense.

The commands have been made 'link-aware' meaning they can now handle Links and objects in linked containers.

This required several changes in the following main files. For each command all options are handled by a single function now (the `move`, `rotate` and `scale` functions). This was the only reasonable solution to correctly handle nested place [... cut]
commit d587e6a9 (2025-01-20): Draft: make move, rotate and scale commands link-aware (#18795)

* Draft: make move, rotate and scale commands link-aware

Fixes #12836.
Fixes #15681.

These changes are presented as a single ('big') PR because the mentioned commands have several similarities. Working on them seperately would have made less sense.

The commands have been made 'link-aware' meaning they can now handle Links and objects in linked containers.

This required several changes in the following main files. For each command all options are handled by a single function now (the `move`, `rotate` and `scale` functions). Th [... cut]
closed 2025-01-20 by yorikvanhavre