FreeCAD/FreeCAD #15684: Stylesheet: some sketcher panels have white-on-light text in FreeCAD Dark theme

depthoffocus, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some of the dropdowns in the Sketcher tasks panel have white text on a light background in the FreeCAD Dark theme:

[image]

This affects the settings dropdowns in the Constraints and Elements panels, and the Recomputation dropdown in the Solver messages panel, but not the two Filters dropdowns.

The background colour of the dropdown appears to be the one uses in the FreeCAD Light theme.

This problem was also in weekly release 38334.

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * fasteners 0.5.25
  * Curves 0.6.42
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-30:
It's been fixed, just missed the creation of the weekly, so you'll have it by Saturday or you can just manually download and overwrite your local copy from https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/Stylesheets/FreeCAD%20Dark.qss

[image]

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38379 (Git)
Build type: Release
Branch: main
Hash: c4bb689a851b16d9a22f085e7406db9dd4cc76dc
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
```

depthoffocus, 2024-07-30:
Oh that's great, I will do that. Thanks! I'll close this.

closed 2024-07-30 by depthoffocus