FreeCAD/FreeCAD #15686: Need for Intuitive and Self-paced Learning Material / Tutorials for Sketcher Workbench

chbhanuvardhan, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For a beginner of FreeCAD, the existing tutorial(s) [https://wiki.freecad.org/Basic_Sketcher_Tutorial] on Sketcher Workbench expects the user to already know about different tools available in Sketcher Workbench. Also, if there are any mistakes made by the learner, there is no real-time feedback on pointing them out and asking the learner to correct the same.

It is desired to have a "Wizard" kind of interface which can be used to learn Sketcher Module, one step at a time.

### Full version info

```shell
FreeCAD 0.21
```

### Subproject(s) affected?

None

### Anything else?

The feedback is from a Teacher's point of view.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-07-30:
You should check this document, it explains a lot: https://owncloud.gwdg.de/index.php/s/eZisrfTTCCjDEd9/download

lucas-bueno777, 2024-07-30:
The tutorial for BIM (even a bit outdated) is an excellent example of this kind of request. It helped a lot the first time I used it.

chbhanuvardhan, 2024-08-01:
Thank you. We shall evaluate the above two suggestions/resources.