FreeCAD/FreeCAD #15691: Assembly: Older project files cannot be loaded without error

wwmayer, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to load the project file Align to selection test Github.FCStd from `https://github.com/FreeCAD/FreeCAD/issues/14576`
gives a lengthy error message:
```
15:17:09  pyException: Traceback (most recent call last):
  File "/home/user/Projects/build_clang/Mod/Assembly/JointObject.py", line 198, in onDocumentRestored
    self.createProperties(joint)
  File "/home/user/Projects/build_clang/Mod/Assembly/JointObject.py", line 202, in createProperties
    self.migrationScript2(joint)
  File "/home/user/Projects/build_clang/Mod/Assembly/JointObject.py", line 465, in migrationScript2
    rootObj, path = UtilsAssembly.getRootPath(obj, part)
  File "/home/user/Projects/build_clang/Mod/Assembly/UtilsAssembly.py", line 282, in getRootPath
    path.append(obj.Tip.Name + ".")
<class 'AttributeError'>: 'str' object has no attribute 'append'
15:17:09  pyException: Traceback (most recent call last):
  File "/home/user/Projects/build_clang/Mod/Assembly/JointObject.py", line 198, in onDocumentRestored
    self.createProperties(joint)
  File "/home/user/Projects/build_clang/Mod/Assembly/JointObject.py", line 202, in createProperties
    self.migrationScript2(joint)
  File "/home/user/Projects/build_clang/Mod/Assembly/JointObject.py", line 465, in migrationScript2
    rootObj, path = UtilsAssembly.getRootPath(obj, part)
  File "/home/user/Projects/build_clang/Mod/Assembly/UtilsAssembly.py", line 282, in getRootPath
    path.append(obj.Tip.Name + ".")
<class 'AttributeError'>: 'str' object has no attribute 'append'
```

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38380 (Git)
Build type: debug
Branch: main
Hash: f03f4ace116348fee143b19a2cc892a85bb11697
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Assembly

### Anything else?

This must be a fairly new error because around one month ago I didn't get it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-30:
@PaddleStroke FYI

wwmayer, 2024-07-31:
Changing the line 282 in UtilsAssembly.py to `path = path + obj.Tip.Name + "."` should fix it.

PaddleStroke, 2024-08-05:
Oh yes sorry for this and thanks for the solution.

closed 2024-08-05 by chennes