FreeCAD/FreeCAD #15692: Techdraw: svg export does retain clipgroup results

RiccaDS, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi, in Techdraw, when one exports to svg a page that contains a clipgroup, the resulting svg displays the view that is in the clipgroup but without clipping it. 
**Steps to reproduce:**
1-Create a new document in freecad
2-Switch to Part Design WB, create an elongated rectangle sketch and pad it with default value. 
3-Switch to Techdraw WB and create a page
4-Select the part that was previously created and the page in the treeview. Click on the "insert view" button in the toolbar and orient the view in such a way that the elongated part of the part is shown in the view. Then click ok.
5-From the Techdraw menu, select Techdraw views -> Insert clip group. 
6-In the treeview, drag and drop the view in the clipgroup. 
7-In the page, select the view within the clipgourp and drag it in such a way that the part is cut by the clipgroup boundaries.
8-export the page to SVG either with the button in the toolbar or with the export feature from the file menu.
9-The svg looks like if there was no clipping.

Example FC file here [clip_error.zip](https://github.com/user-attachments/files/16429133/clip_error.zip)

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.23
  * NikraDAP 2.0.0
  * Render 2024.7.4
  * Assembly4.backup1721345549.1539152 0.50.14 (Disabled)
  * Assembly4 0.50.15
  * fasteners 0.5.24
  * workfeature-macro
  * CurvedShapes 1.0.9
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.2.0
  * Curves 0.6.42
  * OSE3dPrinter 0.1.0
  * Design456 0.0.1
  * kicadStepUpMod 11.2.7
  * QuickMeasure 2022.10.28
  * pcb 6.2023.1
  * Launcher
  * ExplodedAssembly
  * toSketch 1.0.1
  * A2plus 0.4.65
  * Manipulator 1.5.7
  * MakerWorkbench 1.0.1
  * Estimate 0.1.2
  * workfeature
  * FEMbyGEN 2.1.0
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-08-28:
@WandererFan Might be worth waiting to Qt to implement this, rather than doing a work-around. Multiple missing SVG features have been added in Qt 6, and clipGroup seems to be upcomming: https://codereview.qt-project.org/c/qt/qtsvg/+/531968

WandererFan, 2024-10-19:
This one is interesting - https://www.qt.io/blog/qt-svg-not-so-1.2-tiny-any-more

At one point, QtSvg was declared "complete" and no further enhancements were planned.

If QtSvg 6.x will handle clipping svg fills to shape, we can simplify the Svg hatch code quite a bit - no more tiles!

luzpaz, 2024-11-13:
We're looking at Qt6.7 and higher.

FEA-eng, 2025-07-17:
Still relevant

FEA-eng, 2025-11-17:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow)
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

luzpaz, 2026-05-10:
Potentially related TD clip group tickets:
* https://github.com/FreeCAD/FreeCAD/issues/23996
* https://github.com/FreeCAD/FreeCAD/issues/22161