FreeCAD/FreeCAD #15693: AppImage missing Python module "lark", a dependency for ifcopenshell

frodeseverin, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I just dowlnloaded the x86_64 AppImage for FreeCAD 0.21.2. I wanted to open a .dae file in FreeCAD. It opened fine. However, when I read the documentation for [Extra python modules](https://wiki.freecad.org/index.php?title=Extra_python_modules), I figured that these were preinstalled in the AppImage. I decided to try if all the modules were preinstalled, and found that they were, only IfcOpenShell was missing the dependency module "lark".

I proceeded to install the module "lark" as per the instructions for [pyCollada](https://wiki.freecad.org/index.php?title=Extra_python_modules#Linux_AppImages_and_Snaps), replacing "pycollada" with "lark" in the function subprocess.run().

After restarting FreeCAD importing ifcopenshell in the Python console no longer produced the warning about missing dependency function "lark".

Please include the Python module "lark" in the AppImage.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Norwegian Bokmal/Norway (nb_NO)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-30:
ping @adrianinsaval

Syres916, 2024-07-30:
The weekly builds have had it for a significant length of time https://github.com/FreeCAD/FreeCAD-Bundle/commit/a00109d6364bf492cde0b03af6054f119e403603

luzpaz, 2024-07-30:
So, I guess we won't backport ? Then if that's the case this is a "Won't fix" ?

Syres916, 2024-07-31:
@luzpaz I guess you are correct in your assumption but other packaging such as the LibPack will need double checking when test builds are produced before release which I'm making an assumption that will actually happen.

luzpaz, 2024-08-01:
> such as the LibPack

CC @chennes