FreeCAD/FreeCAD #15694: AppImage has Addon manager crash on ObjectsToPython

krushia, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Use AppImage
2. Open addon manager
3. Make sure it is set to show macros
4. Find ObjectsToPython
5. Click it
6. Watch rapidly changing line of text
7. Click back
8. Segfault

`Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3c2a0) [0x7f978b1542a0]
#1  /tmp/.mount_FreeCAlMFQCN/usr/bin/../lib/./libpyside2.cpython-311-x86_64-linux-gnu.so.5.15(+0x11080) [0x7f978a4b9080]
#2  /tmp/.mount_FreeCAlMFQCN/usr/bin/../lib/libQt5Core.so.5(+0xd402c) [0x7f978b60c02c]
#3  /lib64/libc.so.6(+0x8a652) [0x7f978b1a2652]
#4  /lib64/libc.so.6(+0x1068ac) [0x7f978b21e8ac]
`

Everything except the segfault happens with my manual build.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library (Disabled)
  * Curves 0.6.42 (Disabled)
  * A2plus 0.4.65 (Disabled)
  * DynamicData 2.62.0 (Disabled)
  * Assembly4 0.50.15 (Disabled)
  * Defeaturing 1.2.2 (Disabled)
  * dodo 1.0.1 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9 (Disabled)
  * Design456 0.0.1 (Disabled)
  * fasteners 0.5.25
  * SearchBar 1.0.2 (Disabled)
  * DesignSPHysics 0.7.0(15-09-2023) (Disabled)
  * cadquery_module (Disabled)
  * ose-piping (Disabled)
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools (Disabled)
  * Pyramids-and-Polyhedrons (Disabled)
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5 (Disabled)
  * OpenTheme 2024.7.3 (Disabled)
  * Manipulator 1.5.7 (Disabled)
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0 (Disabled)
  * ThreadProfile 1.90.0 (Disabled)
  * freecad.gears 1.2.0 (Disabled)
  * Launcher (Disabled)
  * CommandPanel (Disabled)
  * CubeMenu (Disabled)
  * Reinforcement
  * PieMenu 1.7.5 (Disabled)
  * MeshRemodel 1.9.23 (Disabled)
```

### Subproject(s) affected?

Addon Manager

### Anything else?

Visually it seems to be stick in an endless retry loop, which might be related to https://github.com/FreeCAD/FreeCAD/issues/15059

Not sure what's really going on.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-05:
I cannot reproduce this on Windows and macOS with the latest weekly.
Does this still happen with the latest weekly AppImage?

xtemp09, 2024-08-08:
No crash here.

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38387 (Git)
Build type: Release
Branch: main
Hash: c0c7bd647dc91f8807d8632585b6e6a53a040798
Python 3.10.12, Qt 5.15.14, Coin 4.0.2, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git) AppImage
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

krushia, 2024-08-09:
Still happening. Again also confirming it only crashes with the AppImage, but the manual build still "misbehaves" in the same way except it doesn't crash.

OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git) AppImage
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.42
  * A2plus 0.4.65
  * DynamicData 2.62.0
  * Assembly4 0.50.15
  * Defeaturing 1.2.2
  * dodo 1.0.1 (Disabled)
  * boltsfc 2022.11.5
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.25
  * SearchBar 1.0.2
  * DesignSPHysics 0.7.0(15-09-2023)
  * cadquery_module
  * ose-piping (Disabled)
  * Nodes 0.1.36
  * 3D_Printing_Tools
  * Pyramids-and-Polyhedrons
  * workfeature
  * MnesarcoUtils 0.2.5
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * CommandPanel
  * CubeMenu
  * MeshRemodel 1.9.23

luzpaz, 2024-08-30:
Can you please retest on latest dev again? (not that #15059 was fixed)

Aside: this ticket troubleshooting process would benefit from #15710 being implemented.

krushia, 2024-09-05:
Crash still happens. Seems to be faster now... Also, I should clarify that I have that macro installed already. The crash doesn't happen when trying to install it, but when viewing the page for it in the addon manager.

OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git) AppImage
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.44
  * A2plus 0.4.65
  * DynamicData 2.62.0
  * Assembly4 0.50.15
  * Defeaturing 1.2.2
  * dodo 1.0.1 (Disabled)
  * boltsfc 2022.11.5
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * 3DfindIT 1.2.0
  * CurvedShapes 1.0.9
  * Design456 0.0.1 (Disabled)
  * fasteners 0.5.27
  * SearchBar 1.0.2
  * DesignSPHysics 0.7.0(15-09-2023)
  * cadquery_module
  * ose-piping (Disabled)
  * Nodes 0.1.36
  * 3D_Printing_Tools
  * Pyramids-and-Polyhedrons
  * workfeature
  * MnesarcoUtils 0.2.5
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * CommandPanel
  * CubeMenu
  * MeshRemodel 1.9.23