FreeCAD/FreeCAD #15695: Stylesheet: Preference Checkboxes cut off and deactivated look like partial checked

maxwxyz, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Preferences Checkboxes are cut off.
Disabled checkboxes are not empty but have a `-` in them, like they are when they are partially checked (not all children checked)
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-30:
Seems to be fixed in your build with Qt6 and the latest PR. Will test in the next weekly.

FEA-eng, 2024-08-03:
Built-in Light theme? I still see this in today's weekly.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

MisterMakerNL, 2024-08-03:
This is only in the navigation tab, seem to be a bit of a mess that tab.

closed 2024-08-05 by chennes