FreeCAD/FreeCAD #15696: Preferences: Material page has two nested vertical scrollbars

maxwxyz, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Material preferences have two vertical scrollbars, the outer one only scrolling a few mm which is annoying:
[image]

@kadet1090 @davesrocketshop FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-07-30:
This will depend on the size of your window. In some cases it will be small, but in others it may be large. There's no way to manually resize the inner widget so it relies entirely on a default size. I know of no good solution to this

maxwxyz, 2024-08-03:
It's also on the task dialog where on my screen the material chooser is too large to see it and the launch editor button is hidden:
[image]

luzpaz, 2024-10-23:
> There's no way to manually resize the inner widget so it relies entirely on a default size. I know of no good solution to this

Sounds like a dead-end

FEA-eng, 2026-10-06:
Still relevant

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```