FreeCAD/FreeCAD #15700: Pocket circle hole not done on revolution seam but appears on polar pattern

hasecilu, 2024-07-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to model a bicycle wheel, I made 2 **Pocket** holes to pattern them using a polar array, the problem is that the original instance of the second Pocket hole does not modify the part, instead of a hole there is interruption on the edge that form the face of the revoluted profile (see blue line on image).

[image]

### Full version info

```shell
OS: Arch Linux (Hyprland/hyprland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ThreadProfile 1.90.0
  * FreeCAD-Movie 2024.2.15
  * freecad.gears 1.2.0
  * PieMenu 1.7.3
  * freecad_metal_workbench 0.0.1
  * Ship 2022.4.11
  * OpenTheme 2024.7.3
  * FreeGrid 2.0.0
  * FreeCAD-Reinforcement
  * Defeaturing 1.2.2
  * FreeCAD_SheetMetal 0.4.16
  * FreeCAD-Pyramids-and-Polyhedrons
  * yaml-workspace 0.1.2
  * woodworking 0.21.2.33771
  * Assembly4 0.50.10
  * FastenersWB 0.5.24
  * lattice2 1.0.0
  * Rocket 3.3.0
  * dodo 1.0.1
  * DidYouKnowThat
  * FreeCAD-motion
  * cadquery_module
  * Profiles 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * osh-autodoc-workbench 0.2.3
  * CfdOF 1.27.4
  * A2plus 0.4.64a
  * kicadStepUpMod 11.2.4
  * Nodes 0.1.36
  * Marz 0.1.5
  * frame 0.1.1
  * pcb 6.2023.1
  * FreeCAD_assembly3 0.12.2
  * Render 2024.7.4
```

### Subproject(s) affected?

PartDesign

### Anything else?

I replaced the circle sketch to a square, octagon, Bezier curve and the hole appeared, so it seems only a circle is causing troubles. Maybe an edge case?

Here is my file: [BicycleWheel.zip](https://github.com/user-attachments/files/16434025/BicycleWheel.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FreeWilly2022, 2025-03-23:
I confirmed that there is still an error, although it may be due to the file being corrupted. When I opened the file I got this error:

```
FreeCAD Issues/BicycleWheel.FCStd
17:14:44  Reading failed from embedded file: StringHasher.Table.txt (1068 bytes, 221 bytes compressed)

[Wheel_cleaned.FCStd.txt](https://github.com/user-attachments/files/19411515/Wheel_cleaned.FCStd.txt)

17:14:47  
It is recommended that the user right-click the root of the document and select Mark to recompute.
The user should then click the Refresh button in the main toolbar.
```
I recomputed and there were no errors. But Pocket002 does not actually create a pocket, even though the polar pattern makes pockets everywhere else. Tried changing pocket parameters but still did not work. No errors were reported.

I looked at Sketch005, and it looked corrupted. Sketch005 had a B-spline that looked like garbage, along with a square and octagon that didn't make sense. Editing the sketch resulted in long delays but no errors. After deleting everything I re-made circle. The icon in Tree view indicated not fully constrained even though Solver messages showed fully constrained. Tried Sketcher validation, no errors were found, but did Delete constraints to external geom. and Tree view changed to show fully constrained icon. Hole from Pocket002 is still missing even though the rest of the holes from the polar pattern appear. Saved file as Wheel_clean.FCStd.txt if you want to see the clean file.

OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: arm64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09

pieterhijma, 2025-05-14:
I think it is OCC related. I can confirm. I'll add it to the project.

For future reference: I don't understand why the file is so large. An initial guess is that the problem is the seam line but then on the other hand, it is only the circle that causes problems.