FreeCAD/FreeCAD #15702: The application crashes on mouse click in PartDesignGui

rdv0011, 2024-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCad crashes when the user clicks on the already selected face.
[image]

### Full version info

```shell
[code]
OS: macOS 14.1
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

[light_reflector_basin_copy.FCStd.zip](https://github.com/user-attachments/files/16438004/light_reflector_basin_copy.FCStd.zip)
[freecad_crash.txt](https://github.com/user-attachments/files/16438010/freecad_crash.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-07-31:
Please try with the newest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-07-31:
@JohnOCFII can you reproduce on latest dev ?

luzpaz, 2024-07-31:
@rdv0011 Any chance you can test the dev build 0.22 ? (https://github.com/FreeCAD/FreeCAD/issues/15702#issuecomment-2259978589)

rdv0011, 2024-08-01:
I cannot replicate in the dev. Although once thickness fails the whole dress up feature toolbar becomes disabled.
OS: macOS 14.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)

luzpaz, 2024-08-02:
The thickness bug is separate, I reckon. Can you search the 'thickness' related bugs to see which one is yours ? 

I'm closing this ticket since crash is not reproducible on dev version and fix will not be backported to current stable.

closed 2024-08-02 by luzpaz