FreeCAD/FreeCAD #15703: TNP mitigation Edge-Links  calculation migth still be defect

ElPatoNoEsTu, 2024-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PartDesigen throws an unhandled exception while "recalculate" a fillet after setting "tip"  and inserting another feature.

how to reproduce:
New Body, -> Sketch something -> Pad -> add Fillet -> set Tip to "Pad" -> add another Sketch -> Pad -> Error is thrown in Report Window.

What I notice is that the "Edge-Link" seems to contain some information twice. 

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * DynamicData 2.62.0
  * freecad.gears 1.2.0
  * LCInterlocking
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-01:
Could this be #15487? It also concerns setting the tip to before a fillet.

bgbsww, 2024-08-02:
It looks to be a duplicate.  Note that in 15487 it is pointed out that the errors are correct, but that they can later be removed because opening the fillet dialog takes a guess and reconnects the changed geometry.

The fillet ( any dressup ) has a list of edges that it is referring to.  If you add or edit things before that fillet in the body, those edges necessarily change, and the fillet ends up pointing to edges that have changed.  There isn't really a mechanism to change these during the initial operation, because they won't be detected until after you've completed all the dialogs and are into a recompute of the object.  Once they are detected, there isn't a mechanism to pop up a dialog and so FC informs you by marking those edges are missing.  When you finally enter the fillet dialog, it has a chance to present you with alternative replacement edges and does so.  Or will in main.

Confirmation would be to see if the same thing happens in LS3 and it does.  So this is a duplicate.

ElPatoNoEsTu, 2024-08-02:
Hi, 

why can't the "doing the guesswork which topological object is which" algorithm be triggered to "repair" in case of detection - if the "!" in the Tree occurs - or if the exception is thrown as part of handling the exception ? 

Just because its still unhandled in LS3 imho it doesn't need to stay unhandled in 1.0 

Regards EpNeT

bgbsww, 2024-08-02:
Because you need to confirm with the user, and we are in feature freeze
allowing no UI changes.  It is a good post 1.0 idea that is already under
consideration, but the 1.0 goal is parity.

ElPatoNoEsTu, 2024-08-02:
I would like to ask for adressing this in the Sunday contributor meeting to get more opinions on this.
  
Imho - without fixing this bug TNP can't be stated as mitigated.  Also I woudn't see a confirmation as _new_ feature because its part of the TNP mitigation and just still missing at the very moment to round up TNP mitigaton.  ( analog to the rework which was made last couple of weeks, neccesary to populate some tables when open/saveing in background ).

bgbsww, 2024-08-02:
Go for it if you wish.  Note that I didn't decide we're not making UI changes - that's the rule we're all living under.

The Dressups don't get garbled - they get marked as invalid INSTEAD of getting garbled.  If we left them alone, then sometimes they would get garbled, and that would be a worse state.  This is exactly what the LS3 branch does when it can't guess successfully - and the case of dressups and moving the tip is precisely one of those cases where it often cannot.

If you follow the logic of what you are attempting to do: fillet something, and then go back and change that something and expect the system to guess your intention perfectly in all cases without user intervention, you'll see that it is simply impossible.  Writing a new dialog and adding it to the system at this point with no prior art would be risky.

If you ( or someone else ) wants to write the patch, demonstrate that it works, and then raise that for discussion as an exception that should be included, that would make sense to me.  In the absence of someone providing that, I don't see a conversation being effective, because now you're asking for a new feature that does not and never has existed, and isn't necessary proven to work correctly.

Mitigation does not have to solve every case.  That's why it's mitigating, not solving.

All of that is just my take; maybe someone else will see it differently.

In any case, this is absolutely redundant with #15487 and should be closed.  Conversation can and should continue there about correct approaches.

chennes, 2024-08-02:
OK - thanks for the issue report, @ElPatoNoEsTu, and for the ensuing discussion @bgbsww  -- you are of course welcome to add this to the coming developer meeting agenda, but I will close this as a duplicate of #15487 .

closed 2024-08-02 by chennes