FreeCAD/FreeCAD #15704: Structural members (beams and columns) following a non-ortho axis might have wrong rotation

yorikvanhavre, 2024-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a development of #15393

When creating a BIM beam or column by drawing a 2-point axis or setting its Tool property, and that axis is not a standard ortho direction (X, Y or Z), the beam object might be wrongly rotated around its axis:

![](https://private-user-images.githubusercontent.com/1136856/353465574-4eee7524-21c4-4055-b223-41f9b1840893.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI0MTU2NjUsIm5iZiI6MTcyMjQxNTM2NSwicGF0aCI6Ii8xMTM2ODU2LzM1MzQ2NTU3NC00ZWVlNzUyNC0yMWM0LTQwNTUtYjIyMy00MWY5YjE4NDA4OTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDczMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MzFUMDg0MjQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzQ1YTdkOWUzMGFmZWRkMzZkMDgxNzdhYmFlNzdkM2M3OTI3MTUxZjk2YWEwYmNkZDgzZjI0MDQyYWQ2MGE5MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.F_b-V6XMHommMCnVqsaasylHLnZ-yUzg918b4kD-lj8)

This is probably due to a rotation being calculated between from and to vectors, with no up direction reference: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/ArchStructure.py#L939

### Full version info

```shell
Any FreeCAD version
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-09-27:
Similar issue: #23271.

Roy-043, 2026-11-07:
Should be fixed with #28601.

closed 2026-11-07 by Roy-043