FreeCAD/FreeCAD #15706: "Create Exploded View" crashes FC in Assembly WB

RiccaDS, 2024-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Assembly WB, the "Create Exploded View" button in the toolbar crashes FC. I experience this with two files I attach. 
[assemblytest.zip](https://github.com/user-attachments/files/16442147/assemblytest.zip)
[explodedviewerror2.zip](https://github.com/user-attachments/files/16442153/explodedviewerror2.zip)
Steps to reproduce:
1-Open one of the two files in FC
2-Switch to the Assembly WB, activate the Assembly in the treeview and keep it selected
3-Click the "Create Exploded view" button in the toolbar. FC crashes.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.4.23
  * NikraDAP 2.0.0
  * Render 2024.7.4
  * Assembly4.backup1721345549.1539152 0.50.14 (Disabled)
  * Assembly4 0.50.15
  * fasteners 0.5.24
  * workfeature-macro
  * CurvedShapes 1.0.9
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.2.0
  * Curves 0.6.42
  * OSE3dPrinter 0.1.0
  * Design456 0.0.1
  * kicadStepUpMod 11.2.7
  * QuickMeasure 2022.10.28
  * pcb 6.2023.1
  * Launcher
  * ExplodedAssembly
  * toSketch 1.0.1
  * A2plus 0.4.65
  * Manipulator 1.5.7
  * MakerWorkbench 1.0.1
  * Estimate 0.1.2
  * workfeature
  * FEMbyGEN 2.1.0
```

### Subproject(s) affected?

Assembly

### Anything else?

I have asked in the Forum but with no luck
https://forum.freecad.org/viewtopic.php?p=773507#p773507

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pendelf, 2024-07-31:
It seems to work fine on Windows
https://github.com/user-attachments/assets/e670c0a8-fd03-4a45-a157-09f025052f6d

amreo, 2024-07-31:
It seems to work correctly, I cannot reproduce the bug
```
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
```

amreo, 2024-07-31:
Maybe it is caused by some addon

maxwxyz, 2024-08-05:
I cannot reproduce on macOS and Windows with the latest weekly.
Could you try this with a fresh install (no mods/config files)?

luzpaz, 2024-08-14:
@RiccaDS have you tested with latest dev ?

Pendelf, 2024-08-14:
Everything seems to be working fine

https://github.com/user-attachments/assets/43a67cb4-52e4-4ace-85ba-b36764151ee9

OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.6
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.23

luzpaz, 2024-08-14:
Again, this would be a good usecase for troubleshooting by disabling all addons #15710

RiccaDS, 2024-08-16:
> @RiccaDS have you tested with latest dev ?

I tried it right now in dev 38459. Still crashes. I'm trying to test without addons

RiccaDS, 2024-08-16:
Disabling addons did not have any effect. I also rebuilt the file from scratch in the same exact way in dev38459 appimage, and it keeps crashing. 
I'm not too used to debug FC, any way I can extract a meaningful log that might help?

As soon as I have some time I'll test with a clean install if I understand how to do that.

RiccaDS, 2024-08-16:
It does not crash with this file tough, which is similar but slightly different. What actually changes are the sketch geometries and sketch constraints. 
[assemblytest2.zip](https://github.com/user-attachments/files/16637311/assemblytest2.zip)

pure-machinery, 2024-09-16:
This is happening to me as well.

<details><summary>Console output</summary>
<p>

```
[Abovegame:4023 :0:4023] Caught signal 11 (Segmentation fault: address not mapped to object at address 0x208)
==== backtrace (tid:   4023) ====
 0 0x000000000004d212 ucs_event_set_fd_get()  ???:0
 1 0x000000000004d3dd ucs_event_set_fd_get()  ???:0
 2 0x000000000003d1d0 __sigaction()  ???:0
 3 0x0000000000321b44 SoDragger::addFinishCallback()  ???:0
 4 0x0000000000bad447 Gui::View3DInventorPy::addDraggerCallback()  ???:0
 5 0x0000000000bb897e Py::PythonExtension<Gui::View3DInventorPy>::method_varargs_call_handler()  ???:0
 6 0x0000000000ba4574 Gui::View3DInventorPy::method_varargs_ext_handler()  ???:0
 7 0x00000000001afe1c PyObject_GenericGetAttr()  ???:0
 8 0x0000000000180c7b _PyObject_MakeTpCall()  ???:0
 9 0x0000000000189829 _PyEval_EvalFrameDefault()  ???:0
10 0x0000000000183a46 _PyObject_FastCallDictTstate()  ???:0
11 0x00000000001c0877 PyObject_IsTrue()  ???:0
12 0x0000000000180c28 _PyObject_MakeTpCall()  ???:0
13 0x0000000000189829 _PyEval_EvalFrameDefault()  ???:0
14 0x00000000001e2995 PyObject_CallFinalizer()  ???:0
15 0x00000000001e2498 PyObject_CallFinalizerFromDealloc()  ???:0
16 0x000000000023082d PP_Run_Method()  ???:0
17 0x0000000000159068 Base::InterpreterSingleton::runMethodVoid()  ???:0
18 0x0000000000710b38 Gui::PythonCommand::activated()  ???:0
19 0x00000000007112be Gui::Command::_invoke()  ???:0
20 0x00000000007117e7 Gui::Command::invoke()  ???:0
21 0x00000000001a3397 QSignalMapper::QSignalMapper()  ???:0
22 0x00000000005243c4 QAction::activate()  ???:0
23 0x0000000000210ad5 QFusionStyle::subControlRect()  ???:0
24 0x000000000021d58e QAbstractButton::mouseReleaseEvent()  ???:0
25 0x000000000034750d QToolButton::mouseReleaseEvent()  ???:0
26 0x000000000014d376 QWidget::event()  ???:0
27 0x00000000000fc8cc QApplicationPrivate::notify_helper()  ???:0
28 0x0000000000101e59 QApplication::notify()  ???:0
29 0x00000000006a1c89 Gui::GUIApplication::notify()  ???:0
30 0x0000000000145aa8 QCoreApplication::notifyInternal2()  ???:0
31 0x00000000000f3f9d QApplicationPrivate::sendMouseEvent()  ???:0
32 0x00000000001635c2 QCommonStyle::~QCommonStyle()  ???:0
33 0x0000000000164c20 QCommonStyle::~QCommonStyle()  ???:0
34 0x00000000000fc8cc QApplicationPrivate::notify_helper()  ???:0
35 0x00000000006a1c89 Gui::GUIApplication::notify()  ???:0
36 0x0000000000145aa8 QCoreApplication::notifyInternal2()  ???:0
37 0x0000000000178140 QGuiApplicationPrivate::processMouseEvent()  ???:0
38 0x00000000001e7724 QWindowSystemInterface::sendWindowSystemEvents()  ???:0
39 0x0000000000051b87 QXcbConnection::xi2HandleEvent()  ???:0
40 0x000000000005d299 g_source_query_unix_fd()  ???:0
41 0x00000000000bfec7 g_io_channel_new_file()  ???:0
42 0x000000000005c795 g_main_context_iteration()  ???:0
43 0x00000000003a82bd QEventDispatcherGlib::processEvents()  ???:0
44 0x000000000014ff66 QEventLoop::exec()  ???:0
45 0x000000000014a11d QCoreApplication::exec()  ???:0
46 0x00000000005aa2a6 Gui::Application::runApplication()  ???:0
47 0x0000000000006d0c std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_cold()  ???:0
48 0x0000000000025e08 __libc_init_first()  ???:0
49 0x0000000000025ecc __libc_start_main()  ???:0
50 0x0000000000007105 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_cold()  ???:0
=================================
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7f8d1244c1d0]
#1  0x7f8d11321b44 in SoDragger::addFinishCallback(void (*)(void*, SoDragger*), void*) from /usr/lib/libCoin.so.80+0x4
#2  0x7f8d157ad447 in Gui::View3DInventorPy::addDraggerCallback(Py::Tuple const&) from /usr/lib/freecad/lib/libFreeCADGui.so+0x3e7
#3  0x7f8d157b897e in Py::PythonExtension<Gui::View3DInventorPy>::method_varargs_call_handler(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1ce
#4  0x7f8d157a4574 in Gui::View3DInventorPy::method_varargs_ext_handler(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x14
#5  /usr/lib/libpython3.12.so.1.0(+0x1afe1c) [0x7f8d139afe1c]
#6  /usr/lib/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x2fb) [0x7f8d13980c7b]
#7  /usr/lib/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0xb69) [0x7f8d13989829]
#8  /usr/lib/libpython3.12.so.1.0(_PyObject_FastCallDictTstate+0x1e6) [0x7f8d13983a46]
[... 40 lines cut ...]
```
</p>
</details> 

```
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Installed mods: 
  * BillOfMaterials 0.1.4.2
  * sheetmetal 0.4.25
  * fasteners 0.5.28
  * fcgear 1.0.0
  * OpenTheme 2024.9.1
  * IconThemes
```

> It does not crash with this file tough, which is similar but slightly different. What actually changes are the sketch geometries and sketch constraints. [assemblytest2.zip](https://github.com/user-attachments/files/16637311/assemblytest2.zip)

I get the crash with this file as well.

BoomSaag, 2024-09-28:
I have the same issue with the exploded view button crashing FC.

```
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Kernel: Linux 6.8.0-45-generic
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
```
Here is a very basic assembly I made which repeatedly reproduces the crash.
[testAssembly.zip](https://github.com/user-attachments/files/17174037/testAssembly.zip)

Please see my console output below:

<details>
<summary>Console Output</summary>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7894abc45320]
#1  0x7894ab540ce0 in SoDragger::addFinishCallback(void (*)(void*, SoDragger*), void*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/./libCoin.so.80+0
#2  0x7894af386e55 in Gui::View3DInventorPy::addDraggerCallback(Py::Tuple const&) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so+0x3b5
#3  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so(+0xacda03) [0x7894af4cda03]
#4  0x7894af37f4ef in Gui::View3DInventorPy::method_varargs_ext_handler(_object*, _object*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so+0xf
#5  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7894ae606068]
#6  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7894ae5e5b5c]
#7  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7894ae5f1532]
#8  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(_PyFunction_Vectorcall+0x181) [0x7894ae616b71]
#9  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(+0x21e152) [0x7894ae61e152]
#10  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x24b) [0x7894ae5e5b3b]
#11  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7894ae5f1532]
#12  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(+0x235df4) [0x7894ae635df4]
#13  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libpython3.11.so.1.0(+0x2357b0) [0x7894ae6357b0]
#14  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADBase.so(PP_Run_Method+0xe3) [0x7894ad9e96d3]
#15  0x7894ad93358b in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADBase.so+0x2b
#16  0x7894aefbec23 in Gui::PythonCommand::activated(int) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so+0x53
#17  0x7894aefbf1ef in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so+0x23f
#18  0x7894aefbf5df in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so+0x12f
#19  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7894ac2eb6f2]
#20  0x7894ad16a462 in QAction::triggered(bool) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0x32
#21  0x7894ad16ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0xab
#22  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5(+0x25d29a) [0x7894ad25d29a]
#23  0x7894ad25d3d7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0xe7
#24  0x7894ad34610a in QToolButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0xa
#25  0x7894ad1aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0x178
#26  0x7894ad1703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#27  0x7894ad178262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#28  0x7894aef69079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so+0x99
#29  0x7894ac2b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Core.so.5+0x108
#30  0x7894ad1763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#31  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7894ad1c6e6f]
#32  /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7894ad1ca0c9]
#33  0x7894ad1703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#34  0x7894aef69079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so+0x99
#35  0x7894ac2b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Core.so.5+0x108
#36  0x7894ac9353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Gui.so.5+0x84c
[... cut ...]
#44  0x7894ac2bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libQt5Core.so.5+0x82
#45  0x7894aeec84e1 in Gui::Application::runApplication() from /tmp/.mount_FreeCAjoWyQK/usr/bin/../lib/libFreeCADGui.so+0xa41
#46  /tmp/.mount_FreeCAjoWyQK/usr/bin/freecad(+0x789f) [0x5fb2524d589f]
#47  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7894abc2a1ca]
#48  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7894abc2a28b]
#49  /tmp/.mount_FreeCAjoWyQK/usr/bin/freecad(+0x7c12) [0x5fb2524d5c12]```

</details>

maxwxyz, 2024-09-28:
@PaddleStroke could you look into this?

PaddleStroke, 2024-10-02:
Without being able to reproduce it's quite problematic as I can't test potential fixes. From the console log provided we can see the crash happens in `SoDragger::addFinishCallback`
Which is called by `Gui::View3DInventorPy::addDraggerCallback`
Which is called by the `TaskAssemblyCreateView::init`

```
        self.cbFin = view.addDraggerCallback(
            self.asmDragger, "addFinishCallback", self.draggerFinished
        )
```

@wwmayer I seem to recall that you know about this `Gui::View3DInventorPy::addDraggerCallback` function. Any idea what could cause this bug?

Few ideas I had : 
- `drag` pointer is perhaps invalid? Though it seems `convertSWIGPointerObj` should catch this.
- `method` is perhaps not good either? If `PyCallable_Check `is not catching something wrong?
- Perhaps a problem with the GIL? Though `View3DInventorPy::draggerCallback` already seems to be handling that.

wwmayer, 2024-10-02:
> drag pointer is perhaps invalid? Though it seems convertSWIGPointerObj should catch this.

If this is the case then only if something in pivy has changed. At least in FreeCAD it could additionally be checked if `drag` is a null pointer but if it's a corrupted pointer because of an inappropriate cast then it's probably a pivy issue.

> method is perhaps not good either? If PyCallable_Check is not catching something wrong?

Nope.

> Perhaps a problem with the GIL? Though View3DInventorPy::draggerCallback already seems to be handling that.

Nope.  This function explicitly holds the GIL.

A further possibility is that due to changes in Coin something is wrong with the ref counter of internal nodes. We already had such problems on macOS, FreeBSD and OpenBSD with Coin >= 4.0.1

PaddleStroke, 2024-10-08:
I don't know how to move forward on this issue. Anyone who can reproduce this give this a go ?

pure-machinery, 2024-10-08:
I can try and debug it a bit as much as my ability allows me to.

wwmayer, 2024-10-08:
According to the address of `drag` the conversion from Python to C++ of SoDragger fails. At least it will be easy to avoid the crash by checking drag for null but the other problem is to figure out why the conversion fails.

wwmayer, 2024-10-08:
Can you try the following and report back if it crashes, too:
```
from pivy import coin
g = coin.SoGroup()
Gui.coinRemoveAllChildren(g)
```

```
from pivy import coin
g = coin.SoGroup()
App.newDocument()
Gui.ActiveDocument.ActiveView.dumpNode(g)
```

pure-machinery, 2024-10-09:
None of the propsed snippets crash FreeCAD. 

From the first one i get no response, while from the second i just get some string: 

`'#Inventor V2.1 ascii\n\n\nGroup {\n\n}\n'`

wwmayer, 2024-10-09:
Thanks!

Looking at your gdb output again I just realized that the value of `dragger` is `None`. So, this means if `None` is passed then it crashes.
```
def test():
    pass

Gui.ActiveDocument.ActiveView.addDraggerCallback(None, "addFinishCallback", test)
```

So, the problem is caused by `TaskAssemblyCreateView` where `self.asmDragger` is `None`.

closed 2024-10-11 by chennes

PaddleStroke, 2024-10-11:
@pure-machinery what happens for you with Werner's fix ? I guess it does not crash but then you don't have the dragger that appears right? So the tool cannot be used ? 

If `self.asmDragger` is `None` then we can check if the problems comes from the python binding or if it comes from VPAssembly. If you can compile can you try this change in `ViewProviderAssemblyPyImp `:  
```
PyObject* ViewProviderAssemblyPy::getDragger(PyObject* args)
{
    if (!PyArg_ParseTuple(args, "")) {
        return nullptr;
    }
    Gui::SoFCCSysDragger* asmDragger = getViewProviderAssemblyPtr()->getDragger();

    if(!asmDragger) { Base::Console().Warning(" asm dragger received from VPAssembly is none\n");}

    return Base::Interpreter().createSWIGPointerObj("pivy.coin", "SoDragger *", asmDragger, 0);
}
```

PaddleStroke, 2024-10-11:
Oh I think I know how `soDragger `could be None actually.
How did you activate the assembly? Did you double click on the assembly or did you right click, "activate Object" ?

If you right clicked then menu, I think then soDragger would be None since this was activating the assembly without setting it to edit. (this bug is fixed by this PR : https://github.com/FreeCAD/FreeCAD/pull/17092)

Yes I think it's likely this.

PaddleStroke, 2024-10-11:
I have just tested and confirm that I can crash with exploded view. So the problem was as I said before: if you activate the assembly by right click -> context menu, then the edit was not set, so the dragger is None. So it crashes.

On top of the fix https://github.com/FreeCAD/FreeCAD/pull/17092, I think the Assembly commands isActive should be modified to check if the assembly has the edit or not. To make sure t his kind of issue does not happen again.

RiccaDS, 2024-10-11:
> I have just tested and confirm that I can crash with exploded view. So the problem was as I said before: if you activate the assembly by right click -> context menu, then the edit was not set, so [...]

I confirm, activation of the assembly in tree view through context menu crashes FC after exploded view creation. This does not happen with double click.

pure-machinery, 2024-10-11:
Yup, it seems that when double clicking the program does not crash.