FreeCAD/FreeCAD #15707: Sketcher to SpreadSheet and back - display froze, then crash

JohnOCFII, 2024-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This may be hard to reproduce, but since I had a fresh crash report, thought I'd post it anyway.

Moving between PartDesign/Sketcher and Spreadsheet - I clicked into Spreadsheet to review a value, then clicked back to PartDesign/Sketcher.  The main window stayed in Spreadsheet.  Clicking again, FreeCAD crashed.  I'll watch to see if I can repeat it.

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.7.30
```

### Subproject(s) affected?

Sketcher

### Anything else?

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [40512]
Path:                  /Applications/Other Apps/FreeCAD-38375.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [40506]
User ID:               501

Date/Time:             2024-07-31 12:08:58.7270 -0500
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       9710FFB2-CF0F-4F76-903E-0315501EA95A

Time Awake Since Boot: 1000000 seconds
Time Since Wake:       3147 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Codes:       0x0000000000000001, 0x0000000000000008

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [40512]

VM Region Info: 0x8 is not in any region.  Bytes before following region: 4342890488
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102db4000-102dd4000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-38375.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.2.dylib          	       0x107935200 SoRenderManager::setRenderMode(SoRenderManager::RenderMode) + 12
1   SketcherGui.so                	       0x306d49e9c SketcherGui::ViewProviderSketch::unsetEdit(int) + 332
2   libFreeCADGui.dylib           	       0x104c22bfc Gui::ViewProvider::finishEditing() + 32
3   libFreeCADGui.dylib           	       0x1048161fc Gui::Document::_resetEdit() + 160
4   libFreeCADGui.dylib           	       0x1047a41b4 Gui::Application::setEditDocument(Gui::Document*) + 124
5   libFreeCADGui.dylib           	       0x1048481d4 Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) + 160
6   libpython3.11.dylib           	       0x103847dec cfunction_call + 284
7   libpython3.11.dylib           	       0x1037f0348 _PyObject_MakeTpCall + 332
8   libpython3.11.dylib           	       0x1038f2680 _PyEval_EvalFrameDefault + 46152
9   libpython3.11.dylib           	       0x1038e6268 PyEval_EvalCode + 220
10  libpython3.11.dylib           	       0x10394c3f4 run_mod + 144
11  libpython3.11.dylib           	       0x10394ffa0 PyRun_StringFlags + 112
12  libFreeCADBase.dylib          	       0x1034cbca4 Base::InterpreterSingleton::runString(char const*) + 108
13  libFreeCADGui.dylib           	       0x1048b3440 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 164
14  libFreeCADGui.dylib           	       0x1048b329c Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) + 124
15  SketcherGui.so                	       0x306d844e8 SketcherGui::TaskDlgEditSketch::reject() + 308
16  libFreeCADGui.dylib           	       0x104b9e0fc Gui::TaskView::TaskView::reject() + 100
17  libQt5Core.5.15.13.dylib      	       0x106c29e84 0x106ae4000 + 1334916
18  libQt5Core.5.15.13.dylib      	       0x106c0d094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
19  libQt5Core.5.15.13.dylib      	       0x106c0d094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
20  CoreFoundation                	       0x19b32e46c __CFRunLoopDoSource0 + 176
21  CoreFoundation                	       0x19b32e1dc __CFRunLoopDoSources0 + 244
22  CoreFoundation                	       0x19b32cdc8 __CFRunLoopRun + 828
23  CoreFoundation                	       0x19b32c434 CFRunLoopRunSpecific + 608
24  HIToolbox                     	       0x1a5ad019c RunCurrentEventLoopInMode + 292
25  HIToolbox                     	       0x1a5acfe2c ReceiveNextEventCommon + 220
26  HIToolbox                     	       0x1a5acfd30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
27  AppKit                        	       0x19eb8bd68 _DPSNextEvent + 660
28  AppKit                        	       0x19f381808 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
29  AppKit                        	       0x19eb7f09c -[NSApplication run] + 476
30  libqcocoa.dylib               	       0x1089804d0 0x108954000 + 181456
31  freecad                       	       0x102dbb1cc main + 3356
32  dyld                          	       0x19aec60e0 start + 2360
```

[Pencil Holder Basic.FCStd.zip](https://github.com/user-attachments/files/16445109/Pencil.Holder.Basic.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-01:
I have also had a couple of very odd random _freezes_ when editing dimensions in Sketcher on a Mac, but not reproducibly.

maxwxyz, 2024-08-05:
On what occasions does this happen? Can you reproduce this reliably with a certain file? If so, could you record a macro and upload it?

depthoffocus, 2024-08-05:
In my case it happened when leaving a dimension editor popup and returning to the sketch, but it hasn't happened in the most recent build and I couldn't reproduce it consistently -- I will attempt to reproduce it if it happens again.

I am not a GUI developer but I have a gut feeling that if we do ever reproduce this, it will have something to do with Qt's Mac menu bar handling. 

In the situation I've seen it, closing a dimension popup, a bunch of disabled menu items will be getting re-enabled.

My thinking is the Mac menu behaves oddly throughout FreeCAD -- it sometimes, inconsistently, doesn't appear at all, and that is also _impossible_ to reproduce! macOS global menu bars are presumably a bit of a special case for Qt. And an app totally *freezing* rather than crashing always seems to hint to me something happening a bit further down the stack.

(I would be interested to know from @JohnOCFII if he uses FreeCAD in full screen mode like I do)

JohnOCFII, 2024-08-05:
> (I would be interested to know from @JohnOCFII if he uses FreeCAD in full screen mode like I do)

I don't use full screen mode on my Mac for any apps.  

I've not been able to replicate the issue.  And it is the type of interaction I do fairly often - and it was the first time I'd seen the behavior.  I mainly opened the issue as I had a decent Mac Crash report - hoping that might give a clue, but since it has been a week, and there hasn't been a rash of other similar reports, I'd be OK to close the issue -- I'll give it another day before I close it, but if the maintainers wish to close it, that is OK.

luzpaz, 2024-08-27:
This is macOS specific, right ?

JohnOCFII, 2024-08-27:
> This is macOS specific, right ?

Yes.  I'll close.  I haven't seen the issue in some time.

closed 2024-08-27 by JohnOCFII