FreeCAD/FreeCAD #15709: When one of the model's sketches is successfully edited, other constructions are spontaneously destroyed.

Xanuman, 2024-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I start editing a model that I created yesterday in the same environment.

I change one sketch, and successfully. The rotation part changes shape. And after that, FreeCAD starts to spontaneously destroy other (NOT related to the first change) constructions (extrusions, grooves, etc.)
Any attempt to return everything to its place ends with the error "Recompute Failed"

The changes I made did NOT affect the geometry of other constructions. There are also no connections between different sketches.

Or, for example, the support plane changes its position in space without any intervention.

It's such a pity that FreeCAD, the only free program, is so powerful at first glance, but when you start working with it deeper, you understand how raw, unpredictable and unstable it is.

I have already encountered a similar problem several times in completely different projects.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * AirPlaneDesign 0.4.1
  * Alternate_OpenSCAD 1.0.0
  * Assembly4 0.50.15
  * Beltrami 1.0.8-alpha
  * Cfd
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * Defeaturing 1.2.2
  * dxf-library
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.5.22
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * Help 1.0.3
  * ImportNURBS 1.1.0
  * InventorLoader 1.4.0
  * lattice2 1.0.0
  * offline-documentation
  * Plot 2022.4.17
  * QuickMeasure 2022.10.28
  * Rocket 3.3.0
  * sheetmetal 0.4.18
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-07-31:
This sounds like the toponaming issue. Can you try [the latest weekly version](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)? It has the fix implemented and activated.

luzpaz, 2024-07-31:
Op is describing Toponaming problem. More details about it here: https://www.youtube.com/watch?v=tYNkZ-ry5yA

As pierreporte wrote above, there isn't a complete fix for this, all CAD suffer from it (but at this moment mitigate it better than how FreeCAD does). But toponaming improvement is happening as we speak. 

Closing ticket.

closed 2024-07-31 by luzpaz