FreeCAD/FreeCAD #15710: It is not possible to determine which add-on is causing the error so that this add-on can be disabled.

Xanuman, 2024-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After installing several (more than a dozen) add-ons at once, an error appears when launching FreeCAD

A lot of Unicode symbols fall out and something like this:

20:33:13 environment var 'WM_PROJECT_VERSION' is not defined, fallback to hard-coded (5, 0)
20:33:13 During initialization the error "(unicode error) 'unicodeescape' codec can't decode bytes in position 2-3: truncated \UXXXXXXXX escape (<string>, line 1)" occurred in C:\Users\Master\AppData\Roaming\FreeCAD\Mod\Cfd\InitGui.py
20:33:13 Please look into the log file for further information

To somehow bring all this back to normal, you need to disable all installed add-ons and use the elimination method to find which add-on caused the error.
Maybe it's worth coming up with some kind of software method for self-diagnosis of FreeCAD when initializing add-ons, so that the user could quickly identify a buggy add-on and disable it?

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * AirPlaneDesign 0.4.1
  * Alternate_OpenSCAD 1.0.0
  * Assembly4 0.50.15
  * Beltrami 1.0.8-alpha
  * Cfd
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * Defeaturing 1.2.2
  * dxf-library
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.5.22
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * Help 1.0.3
  * ImportNURBS 1.1.0
  * InventorLoader 1.4.0
  * lattice2 1.0.0
  * offline-documentation
  * Plot 2022.4.17
  * QuickMeasure 2022.10.28
  * Rocket 3.3.0
  * sheetmetal 0.4.18
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-08-01:
I'd go further and say **all** messages should be prefixed with something that identifies the function caller. This includes those in the core, which I'm not sure how to do in C++ (but it must be possible?).

luzpaz, 2024-08-01:
This request is related to #6106 where we have a 'safe mode' type of function that allows the user to choose which specific (or all) addon(s) to disable when running an instance of FreeCAD.

luzpaz, 2024-08-04:
We could also in the meantime, create a flag that disables all plugins when invoked from CLI/

luzpaz, 2024-08-08:
I've renamed ticket. 

Idea: add a flag to the CLI that triggers the addon manager to disable all addons when FreeCAD is invoked. This would be really helpful for triage and bug tracking.

luzpaz, 2024-08-09:
@chennes do you mind weighing in regarding on what infrastructure could be used to disable all addons upon invoking freecad ?

chennes, 2024-08-09:
The existence of a file called "ADDON_DISABLED" in the top of an Addon directory prevents FreeCAD from loading it, so anything that batch-creates those files would work. Or that strategy could be used at the top-level of the Mod directory structured to globally disable all of them (though that would be a new feature, it doesn't exist).

luzpaz, 2024-08-09:
Thanks for weighing in, I did a search in the code to see how to call the function to create [ADDON_DISABLED](https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD%20ADDON_DISABLED&type=code) but couldn't quite grok it. 

IMHO, I think option 2 is better (ADDON_DISABLED in the top directory of Mod/) for now. In the future it would be sweet to have better granular control and that's where option 1 (ADDON_DISABLED in top directory of a specific addon or addons).

I think this would be a really important feature for troubleshooting and triage moving forward.

spanner888, 2024-08-09:
The top level trigger to disable all Addons could be a file marker as suggested, or even just a User Preference. If it is a file marker, it really should have a different name to the existing per addon ADDON_DISABLED files, so there is no chance of changing any of the User's ADDON_DISABLED files.

At the code level of how ADDON_DISABLED currently works, after several reviews, my grokking has got as far as noting that:
1. All internal and external/Addons use the same load/init process
2.There are two similar process, one for non-gui init (App) and one for Gui init of each WB/addon
3. Builtins have their init run first, then remaining Mod dirs are processed for Addons. Look at FreeCAD.log and you can see the order and locations.
4. ...from memory 3 types of actual init depending on a hierachy of choices about how that WB/addon was written. This part of the code is not overly complex, but when you need to remember which Addon uses which init type for debugging and App vs Gui, you (well me) really need to be on your toes. There are often log messages that appear to show an issue with a particular WB/Addon, even built in ones, that are really just one of the different init type processes saying that this WB/Addon does not fit this init method, ignoring/skipping it.

GAZ082, 2024-08-30:
I'd go a bit further and would also suggest not only disable addons, but also not using non-default config settings.

This can be either using a second flag or embedding all into one.

luzpaz, 2024-09-12:
@hyarion don't mean to be too forward but would you be interested in considering this as a feature you'd implement ?

hyarion, 2024-09-16:
@luzpaz sure thing, I can look into it.

hyarion, 2024-09-16:
@luzpaz there you go: https://github.com/FreeCAD/FreeCAD/pull/16586

I added support for both disabling addons from command line using `--disable-addon <addon>` (multiple can be added to the command line) and by placing a `ALL_ADDONS_DISABLED` in the mod dir you want to disable addons in.

I tested to add a command line option to disable _all_ addons as suggested in this issue, but that will disable all mods, including start, part, tux etc. The seam would happen if you create stop files in them all.