FreeCAD/FreeCAD #15713: Material: materials-editor.ui not installed in the correct location

nickoe, 2024-07-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was trying to build freecad via the [freecad-git package on ArchLinux AUR](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=freecad-git).  

I ran `namcap` on the generated package, and noticed that materials-editor.ui was copied/installed outside the the pkg dir aka the DESTDIR location for the cmake install target.

This appears to be caused by a type caused by the addition of the ui file in commit 810f34ea11dca01ee833e7c625f87feac39ae2c1

I will submit a PR for this.

### Full version info

```shell
Self build on commit c5408de52591f4a1dfebe14a2a3b589af58d1f3d (origin/main, origin/HEAD)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-31:
Looks like a dupe of #15571

nickoe, 2024-07-31:
@luzpaz  Oh, yes indeed :S  I have created the PR for the ui file

luzpaz, 2024-07-31:
Can you notate the other ticket, pls? TIA :pray:

nickoe, 2024-07-31:
Yeah, so I just compared with the [CMakeLists.txt in the FEM/Gui](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Fem/Gui/CMakeLists.txt) dir, and it does it a bit differently.

Maybe @adrianinsaval 's fix is more correct. https://github.com/FreeCAD/FreeCAD/pull/15565/commits/b9a2735306461973cbb4520c1b53563e8849ec53

nickoe, 2024-07-31:
> Can you notate the other ticket, pls? TIA 🙏

@luzpaz  I updated my commit to include the other issue number, see commit:
c3b384f43b54779d683c20300fdda4623e7f6521

maxwxyz, 2024-08-26:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/15571

closed 2024-08-26 by maxwxyz