FreeCAD/FreeCAD #15717: Part: Sketcher_NewSketch tool does not have fixed place in Part toolbar

Roy-043, 2024-08-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Start-up WB is PartDesign.
2. Switch to the Part WB.
3. The Sketcher_NewSketch tool comes 1st in the "Part tools" toolbar.
4. Switch back to PartDesign and then to Part again.
5. Result: Sketcher_NewSketch comes last in the mentioned toolbar.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-01:
@FEA-eng in your PR https://github.com/FreeCAD/FreeCAD/pull/14318 you've mentioned the same behavior as this issue.
@kadet1090 @Roy-043 did https://github.com/FreeCAD/FreeCAD/pull/15933 maybe solve this too?

Roy-043, 2024-09-02:
Still present in this version:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

luzpaz, 2024-09-30:
@mwganson any chance you can tackle this ?

mwganson, 2024-09-30:
It appears already to be fixed in the latest main branch.  It's always the last time in the toolbar for me.

FEA-eng, 2024-09-30:
> It appears already to be fixed in the latest main branch. It's always the last time in the toolbar for me.

I still have the issue. The button should be first as it is initially.

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git)
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

mwganson, 2024-09-30:
I am able to reproduce the issue now.  I will look into it.

closed 2024-10-07 by chennes