FreeCAD/FreeCAD #15718: assemblyWB: double clickinga perpendicular joint hides it

Panda631, 2024-08-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Have a perdendiculat joint in a assembly. Basically it works as defined.

[image]

double click to one of them deletes it.

[image]

The opened form show the same.

[image]

Selecting one of its faces (without closing the form) restore the joint.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.21
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.1.23
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Anything else?

[assemblyTest.zip](https://github.com/user-attachments/files/16456674/assemblyTest.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-08-06:
This is surprising. Double clicking in the tree works, double clicking in the view other joints work. But not perpendicular in the view.

PaddleStroke, 2024-08-06:
In your sample file, the perpendicular joint is redundant with the slider joint. If you remove the slider joint, then the bad behavior disappear: you can then double click the perpendicular joint correctly.

Even though it still needs to be fixed I think.

PaddleStroke, 2024-08-06:
Ok I found what's happening. The release of the second click, that happens after the joint taskbox is loaded, is unselecting the edge, thus removing the edge from the selection of the joint

closed 2024-08-12 by chennes