FreeCAD/FreeCAD #15719: [Feature Request] PD Mirror could have an offset to the mirrored position.

leoheck, 2024-08-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am wondering here, Part Design Mirror could have an Offset to the mirrored position.

I understand it can be done with the Linear Pattern but sometimes a mirror is easier.

Here is the current mirror operation just as a reference to discuss it.
[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.23
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-01:
I think it would be good if the mirror options matched the Part options, which let you set a point. 

In the case you're talking about, I imagine the way forward is to add a datum plane, offset from the plane or edge you're mirroring on, and use that.

(What would be really nice is if the Part Design Mirror option would let you directly select a datum plane in as many places where it offers up the base planes.)

leoheck, 2024-08-01:
I think the plane can be custom already. Unfortunately my screenshot did not capture that. But an offset from any plane would be nice when you want mirrored copy of the feature that cannot be placed right on the mirrored distance.

depthoffocus, 2024-08-01:
Yes -- it can be custom, you have to pick "select reference" first -- I'd just like to see the datum planes be more directly usable like base planes.