FreeCAD/FreeCAD #15720: Keyboard navigation is broken on many windows.

leoheck, 2024-08-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Many popup windows have issues when navigating with the keyboard.

This is one example of those windows.
[image]

In this example.
The Name field starts selected.
If I hit tab (or shift+tab) I can never go back to the Name field.

If Group is selected, shift+tab cycles throw Group and Type only.

If Type is selected (or any other item under it) I cannot go back to Name and Type by hitting tab.

All the fields should be able to be reached with Tab and Shit+Tab. Preferably in the other, they appear.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.23
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-05:
@pieterhijma @kadet1090 FYI

pieterhijma, 2024-08-05:
@leoheck, thanks for reporting. 

I'm aware that this is an issue for the VarSet dialog. I didn't know it was an issue for other windows. I've been having trouble to get this right and kind-of decided for myself that I will address it as soon as we are dropping Qt 5 for Qt 6. 

Could you please report this on a different window/dialog as well? I can then take a look at whether it is a similar problem as I am having with the VarSet dialog.

leoheck, 2024-08-05:
I think it happens mostly on 3d party workbenches since I don't have any example of the core Freecad on top of my head. But for that, it would be enough to have a guide on how to build windows and widgets focusing the usability since anyone can contribute with workbenches and tools. One example of that would be the Asm4 > Edit Placement window. 

However, while writing this, I saw that Preferences > General tab has a mess of navigation too.
If you select the dropdown list of the `Language`, when you hit Tab, it moves the focus to the `Size of recent file list` which is in the middle of the window.

leoheck, 2024-08-16:
Other examples:

`Part Design > Pad`: I cannot access the `Reversed`, but I can do it on `Part Design > Revolution`, for instance.
[image]

`Part Design > Additive Loft`:
No default widget is selected, so it is hard (requires a lot of key hits) to get the focus to a known place when editing this feature.

`Part Design > Additive Pipe`: the same as `Part Design > Additive Loft`, also the same as in `Loft` and `Subtractive Pipe` 
I can't also reach the `Add Edge`

`Part Design > Pocket`: Cannot reach the checkbox `Reversed` 

`Part Design > Hole`: same issue as in `Part Design > Additive Loft`:
I can't also reach the `Model Thread`
From `Clearance` it jumps to the `Thread Depth` (crazy)
Cannot access `Direction`.

I am tired already... All the Part Design Tasks are poorly organized. They need a bit of love

```
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36438 (Git) AppImage
```

xtemp09, 2024-08-16:
This is called _Tab order_.[^1] There is also _Keyboard Focus_ (labels must not have one).[^2]

Good thing is it can be fixed practically in a matter of two seconds. =)

[^1]: https://doc.qt.io/qt-6/designer-tab-order.html
[^2]: https://doc.qt.io/qt-6/focus.html

kadet1090, 2024-08-16:
I've done that for few preferences pages - but it takes time as you have to do it per every dialog.

luzpaz, 2024-08-17:
There is a tab order script that @chennes made, I think? Right?