FreeCAD/FreeCAD #15722: Confusing window title

leoheck, 2024-08-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Title of the VarSet window is kind of confusing also, it uses the Name of the object not the Label of it.
[image]

I suggest something like this:

- Title: `Variable Set Settings`
- 1st row: `File: This is the name of the filename of this set`
- 2nd row: `VarSet: Lable of this set`

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.23
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-08-01:
Can we have Text/String as a variable? I could not find it.

leoheck, 2024-08-01:
Why? Can you convert forbidden characters to something else instead of throwing this popup every time?
[image]

maxwxyz, 2024-08-01:
@pieterhijma FYI

Roy-043, 2024-08-01:
> IT CLEANS THE WHOLE NAME FIELD!!

See #15551.

pieterhijma, 2024-08-04:
> Why? Can you convert forbidden characters to something else instead of throwing this popup every time? ![image](https://private-user-images.githubusercontent.com/1277920/354305500-60ace2db-faef-4b88-af9b-feaa5f66416a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI3NzgxNzYsIm5iZiI6MTcyMjc3Nzg3NiwicGF0aCI6Ii8xMjc3OTIwLzM1NDMwNTUwMC02MGFjZTJkYi1mYWVmLTRiODgtYWY5Yi1mZWFhNWY2NjQxNmEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgwNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MDRUMTMyNDM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDVmMzQ0ZmIxMjRiOWY0MDllMmQxZWUxM2M2ODYwNjdmZTE2OGViNWM0MTk0YmFiOGE5YWRhMjY2NjZhODNmNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.bORkszrhQQsh3rycg4jYy1jBStoM3gf9LvcggIlZQQE)
> 
> When I hit ok on this IT CLEANS THE WHOLE NAME FIELD!! Come on! Keep whatever the user put, he will fix it if it is required. BUT WHEN HITTING OK, not when leaving the field, PLEASE! This is horrible to use.

This should be fixed by https://github.com/FreeCAD/FreeCAD/pull/15752.

pieterhijma, 2024-08-04:
> > App::PropertyString
> 
> Nice, thank you, @Roy-043. Can we have something easier for the end user? This list is huge. The input field could allow adding text, so I type Str.. and it shows items with this substring in it.

I assume you are talking about the large list of types. Actually, the combo box already allows inserting text to find your types.

leoheck, 2024-08-08:
I cannot add a var from the VarSet using the VarSet Label, it uses the Name of the varset. And this doesn't seem right. 
At least the expression editor does not find if by the Label.

pieterhijma, 2025-08-27:
It is indeed a bit confusing if users are used to the label of the VarSet. In light of #14633, I will simply adopt the strategy of the "old dialog" that has as title "Add Property". In the GUI it is actually quite clear to which document object you are adding properties, so I think this is a good solution.

leoheck, 2025-08-28:
Why does the filename have to be in the title of this window?

This is the default (minimum) window size/width. 

[image]

The filename helps nothing since the file name is the one that is already open in front of the person.
On Linux/Gnome, this pop-up window prevents me from changing the tabs behind it. 

This filename also increases the length of the window and does not show the varset label/name reference by default.

It should show at least which varset is being used clearly (`Label` or `Label#Name` if they differ.)
Can we have this info inside this screen? It would help a lot.

leoheck, 2025-08-28:
@pieterhijma, there is just a single thing I would like to check.
When creating a new VarSet, it creates the object VarSet004 in the objects tree.
And then Fc launches the pop-up window, but it does not show clearly which VarSet object is being edited. 
Can the new object be selected in the tree when it is created? 
Also, this creation window could have a field to label de new VarSet.

[image]

When you edit an existing varset, the selected one is highlighted, so we know where things are going to be. See:

[image]

```
FreeCAD/build on  reconcile-add-prop-dialogs [!] took 2m 18.0s …
FreeCAD 1.1.0, Libs: 1.1.0devR43071 (Git)
```

leoheck, 2025-08-28:
If the new object starts selected... then it is good, does not need any reference inside nor an input to set the label, since the user can quickly rename it by hitting f2 after closing the window.

PR #23426 (2025-09-13): Gui: Reconcile Add Property Dialogs
This PR removes the old "Add Property" dialog in favor of the "Add Property" dialog of VarSets.

The VarSet dialog is now generic and works for any document object or document. In documents, you don't have to select a property any longer to add a new property.

Please note that other than the old "Add Property" dialog, this dialog does not support adding properties to multiple document objects. Since this was an obscure feature (see https://github.com/FreeCAD/FreeCAD/issues/23424) and allowing adding values directly makes the logic more complicated, I believ [... cut]