FreeCAD/FreeCAD #15724: Segfault when drawing cosmetic line on large-scale drawing

ion201, 2024-08-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Minimum reproducible example (it's just a sketch with a 100 ft x 60 ft rectangle, and a techdraw page with that sketch as a view).

[techdraw_crash_cosmetic_line.zip](https://github.com/user-attachments/files/16460555/techdraw_crash_cosmetic_line.zip)

1. Open attached project and focus the techdraw page.
2. Select two opposite corners like shown and click "Add a cosmetic line through two points" from the techdraw toolbar. Freecad segfaults after hitting "OK".
3. Note the exact same process was already done to make a line in the smaller rectangle, it only fails in the large rectangle which is 60'x100'. Arbitrary dimensions, I did not test what minimum line length makes it crash.

[image]

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3cae0) [0x744909650ae0]
#1  0x74482b6e00e9 in TechDraw::BaseGeom::findEndPoints() from /usr/lib/freecad/lib/TechDraw.so+0x19
#2  0x74482b6e0297 in TechDraw::BaseGeom::getStartPoint() from /usr/lib/freecad/lib/TechDraw.so+0x27
#3  0x74482b705531 in TechDraw::CosmeticEdge::CosmeticEdge(std::shared_ptr<TechDraw::BaseGeom>) from /usr/lib/freecad/lib/TechDraw.so+0x151
#4  0x74482b70574f in TechDraw::CosmeticEdge::CosmeticEdge(TopoDS_Edge) from /usr/lib/freecad/lib/TechDraw.so+0x6f
#5  0x74482b705854 in TechDraw::CosmeticEdge::CosmeticEdge(Base::Vector3<double>, Base::Vector3<double>) from /usr/lib/freecad/lib/TechDraw.so+0x64
#6  0x74482b729ad2 in TechDraw::CosmeticExtension::addCosmeticEdge[abi:cxx11](Base::Vector3<double>, Base::Vector3<double>) from /usr/lib/freecad/lib/TechDraw.so+0xf2
#7  0x74482bb13920 in TechDrawGui::TaskCosmeticLine::createCosmeticLine() from /usr/lib/freecad/lib/TechDrawGui.so+0x1c0
#8  0x74482bb13dbb in TechDrawGui::TaskCosmeticLine::accept() from /usr/lib/freecad/lib/TechDrawGui.so+0x3b
#9  0x74482bb0f245 in TechDrawGui::TaskDlgCosmeticLine::accept() from /usr/lib/freecad/lib/TechDrawGui.so+0x15
#10  0x74490d29136a in Gui::TaskView::TaskView::accept() from /usr/lib/freecad/lib/libFreeCADGui.so+0x7a
#11  /usr/lib/libQt6Core.so.6(+0x1a16f7) [0x744909da16f7]
#12  /usr/lib/libQt6Widgets.so.6(+0x2e3333) [0x74490aee3333]
#13  /usr/lib/libQt6Core.so.6(+0x1a16f7) [0x744909da16f7]
#14  /usr/lib/libQt6Widgets.so.6(+0x20e24b) [0x74490ae0e24b]
#15  /usr/lib/libQt6Widgets.so.6(+0x20e748) [0x74490ae0e748]
#16  0x74490ae1b12e in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0xde
#17  0x74490ad4c15e in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xc0e
#18  0x74490acfc51c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#19  0x74490ad019c1 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x10a1
#20  0x74490ce4c1b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#21  0x744909d44d28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#22  0x74490acf3e2d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x21d
#23  /usr/lib/libQt6Widgets.so.6(+0x161fb2) [0x74490ad61fb2]
#24  /usr/lib/libQt6Widgets.so.6(+0x1635f0) [0x74490ad635f0]
#25  0x74490acfc51c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#26  0x74490ce4c1b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#27  0x744909d44d28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#28  0x74490a376df0 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x8d0
#29  0x74490a3e5d64 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x304
#30  /usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x51637) [0x7449044ef637]
#31  /usr/lib/libglib-2.0.so.0(+0x5ca89) [0x7449085d0a89]
#32  /usr/lib/libglib-2.0.so.0(+0xbe9b7) [0x7449086329b7]
#33  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7449085cff95]
#34  0x744909fa2cbd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#35  0x744909d4f01e in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x19e
#36  0x744909d4934d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#37  0x74490cd5d1a7 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x2497
#38  freecad(+0x7cdf) [0x58f158740cdf]
#39  /usr/lib/libc.so.6(+0x25c88) [0x744909639c88]
#40  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x744909639d4c]
#41  freecad(+0x80d5) [0x58f1587410d5]
```

### Full version info

```shell
[code]
OS: Arch Linux (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Manipulator 1.5.6
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-01:
Can you reproduce on dev build ?

WandererFan, 2024-08-02:
As a workaround, go the advanced tab in Preferences and check "Allow crazy edges".   It should work after that.

The diagonal is > 10m, so it falls into the "crazy edge" category.  The test is supposed to be checking the size on the page (a 10m piece of paper seems unlikely), but it looks like it is testing the 1:1 edge length.

luzpaz, 2024-09-16:
@ion201 please test weekly build to confirm.

luzpaz, 2024-09-16:
@WandererFan maybe consider adding 'Fixes #15724` in the commit summary in the future?

closed 2024-09-16 by luzpaz

ion201, 2024-09-16:
Confirmed, issue is resolved