FreeCAD/FreeCAD #15726: TechDraw: Balloon Annotations on section views can have wrong reference

johnsonm, 2024-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Rename [Cap.zip](https://github.com/user-attachments/files/16464073/Cap.zip) to Cap.FCStd, then open the drawing. Control-click to drag the "Thread M12x1.5" annotation with its arrow; the tip does not follow the text. It looks like X and Y might be reversed.

Similarly, trying to attach an annotation to the selected point on the left hand side of this section view:

[image]

...puts the arrow of the annotation elsewhere:

[image]

It is also difficult to position the leader lines appropriately, even after using the Origin X and Origin Y manually to move the location of the tip, the lines go through the drawing when I would not expect them to. This seems related?

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38383 (Git)
Build type: Unknown
Branch: main
Hash: c5408de52591f4a1dfebe14a2a3b589af58d1f3d
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.5.24
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-08-02:
Note to self:  this happens on any rotated base view, not just sections.  the drag operation needs to be smarter in interpreting the mouse move positions.

johnsonm, 2024-08-07:
@luzpaz since @WandererFan has confirmed it, perhaps no longer "needs confirmation"?

closed 2024-09-02 by yorikvanhavre

johnsonm, 2024-09-02:
Sadly, this was not completely fixed. I'll open a new issue.

commit e88b1b49 (2024-09-02): [TD]fix balloon drag on rotated view (fix #15726)