FreeCAD/FreeCAD #15727: Not all constraints are copied when copying part of the sketch

nazar-pc, 2024-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to find an existing issue like this, but didn't succeed, apologize in advance if this is a duplicate.

I basically tried to copy a constrained rectangle (Ctrl+C/Ctrl+V) within the same sketch and only horizontal/vertical constraints were preserved for some reason, which was unexpected and confusing:

https://github.com/user-attachments/assets/9ae177df-dd71-45ec-bc45-23b335fc04fc

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-08-02:
Related to #15653 and #15579.

kaiwas, 2024-08-02:
In the topics by reference, "expressions" are lost.
In this example, just constraints are lost.
I think this is not exactly a duplicate, although the problems are clearly similar.

Something bad happens when copying )
When copying, the user expects to receive an analogue of the geometry with all the previously specified parameters, but receives only a "visually" similar analogue.

FEA-eng, 2026-01-21:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260113 (Git shallow)
Build date: 2026/01/13 22:48:50
Build type: Release
Branch: (HEAD detached at 4cc0a1b)
Hash: 4cc0a1b718735be10261d936066cf1382dd51884
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

PaddleStroke, 2026-09-07:
I see. This is not a bug it's a feature request.

What the copy tool does is that it will copy only constraints that concern geometry that are selected.

In your example the equal constraints are to geometries that are not selected. So it is normal that they are not included in the copy.

If you pasted your copied selection to another sketch, what would happen? The equalities would be unvalid.

This can probably be closed. Or else you need to define precisely the feature request, like what you would like to see happen so that it can be discussed

kaiwas, 2026-09-07:
> If you pasted your copied selection to another sketch, what would happen? The equalities would be unvalid.

I agree that copy-pasting within one sketch is one thing, while copying and pasting into another sketch is a completely different task (I think it's rarer).

But I don't believe that a programmer's bright mind can't take these points into account.
1) Copy-pasting within one sketch copies the statement, not the value (the expression result).
2) Copying and pasting into another sketch pastes the expression result. BUT! The user must be informed of this, otherwise they (like me before) will think they pasted the expression, not the result.

kaiwas, 2026-09-07:
> Or else you need to define precisely the feature request, like what you would like to see happen so that it can be discussed

Just imagine you know nothing about the program's inner workings. At the very least, you should be informed that it's not the statement that's being copied.

I'll try to formulate the proposal more clearly.
But in any case, silently converting an expression into a result when copying is not a good idea. The user doesn't know about it.

nazar-pc, 2026-09-07:
> In your example the equal constraints are to geometries that are not selected. So it is normal that they are not included in the copy.
> 
> If you pasted your copied selection to another sketch, what would happen? The equalities would be unvalid.

As a software engineer I understand what you're saying. As end user this makes little sense to me and I would still argue it is a bug or at very least an unexpected behavior.

When I fully select a constrained piece of geometry and create a copy, I expect the copy to have the same constraints, even if some of the constraints are not internal, but reference other parts of the sketch.

When pasting into a different sketch or document, I'd expect it to still succeed, but get a notification/warning that some constraints (probably a list of them or something) were not transferred because the dependencies are missing/not possible to reference here.

PaddleStroke, 2026-09-08:
> As a software engineer I understand what you're saying. As end user this makes little sense to me and I would still argue it is a bug or at very least an unexpected behavior.

Which is why I said that it is a valid feature request and that you can rephrase the issue to better describe what you want to be supported. :)