FreeCAD/FreeCAD #15730: Updates to sketch can't be seen outside sketch editor

pestalella, 2024-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Created a sketch and imported a couple of segments from a ShapeBinder. Padded it to create a wall. Saw that it wasn't exactly what I wanted and went to the Sketch Editor to move some points. Fully constrained, I closed the Sketch Editor. I expected the pad to reflect the changes but that did not happen. Highlighted the sketch and saw that it was still the unchanged sketch. Went to the Sketch Editor and the changes were there. No amount of Recompute changed anything.
[image]
[image]

### Full version info

```shell
[code]
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.7
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pestalella, 2024-08-02:
Deleting the original pad and creating a new one used the updated sketch. Some intermediate update is missing, apparently.

pestalella, 2024-08-02:
When one of those inconsistencies appears, FreeCAD is unable to export a usable .STL.

luzpaz, 2024-08-03:
Can you reproduce on latest dev build ?

pestalella, 2024-08-03:
On the latest DEV build I see a few warnings:

> 11:00:44  <Exception> Attacher.cpp(876): AttachEngine3D: subshape not found CopyFillet002.
> 11:00:44  DatumPlane: AttachEngine3D: subshape not found CopyFillet002.

And there are red signs on the Model tree view, but at least I can now export to STL what I want.

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git) AppImage
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.7

```

luzpaz, 2024-08-03:
Can you upload the example file (zip it and drag-drop into github textbox to attach it)

pestalella, 2024-08-03:
With the DEV version I managed to reattach the DatumPlanes to the right face and fixed the issue. The original file, without the fixes, is  [LlumLlit.zip](https://github.com/user-attachments/files/16480985/LlumLlit.zip)
What bothered me was that I couldn't see what was going on with the released build. And the error messages on ther DEV version were just hints at what was happening, not something easy to debug for a newbie like me.

maxwxyz, 2024-08-05:
So what issue is left on the dev builds?
Could you record a macro of what you are doing and upload it here?

pestalella, 2024-08-05:
In the DEV build I see the problem stated in the summary.
[no_pad_update.zip](https://github.com/user-attachments/files/16500323/no_pad_update.zip)

luzpaz, 2025-07-28:
Can't reproduce crash but looks like the errors in the report console are similar to:   
* #15392
* #19597