FreeCAD/FreeCAD #15731: Add thread pitch value in the hole tool panel

RayTracedObserver, 2024-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Problem : 

When creating a threaded hole using the hole tool from part design, one has to choose between regular or fine profiles within the hole parameters tasks panel. This setting combined with the hole size automatically determines the thread pitch. However in order to know for sure which thread pitch is being modeled one has to look up the standard values for thread pitch values for coarse and fine (which are called regular and fine in freecad) either online or in a book or wherever the reference is written down.

Solution : 

This could be much improved by displaying the numerical value of the thread pitch within the hole parameter panel.

Thank you for reading 🌞♥️

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-03:
@RayTracedObserver thanks for the report. Is there a way you can wip up a mockup of how this would look like in the UI ?

RayTracedObserver, 2024-08-03:
Yes it could perhaps look like this : 
[image]

Please note that I chose a totally random and incorrect thread pitch value for this mockup.

Thank you !

@luzpaz

pierreporte, 2024-08-25:
Not only should the pitch value be displayed, but it should also be set by the user. There are cases where it is necessary to make specific threads.

RayTracedObserver, 2024-08-28:
@pierreporte Well, it is set by the user when they choose the ISO metric regular or fine profile. But that's a bit counterintuitive perhaps compared to having a menu where we select the thread pitch value. At least if it would display the thread pitch it would be a step in the right direction ! Or maybe you meant we should be able to set the thread pitch to any value ?

pierreporte, 2024-08-28:
Yes I mean that the user should be able to set an arbitrary thread pitch for any diameter.

RayTracedObserver, 2024-08-28:
> Yes I mean that the user should be able to set an arbitrary thread pitch for any diameter.

Ah yes that would be great.

RayTracedObserver, 2024-08-28:
However I also like that the standard threads already come as presets. Ideally we should be able to choose from a list of ISO presets or to enter an arbitrary value as an option.

pierreporte, 2024-08-28:
It’s actually essential, hence the P1 priority. Standard threads available are a must as well so they won’t go away.

RayTracedObserver, 2024-08-28:
> It’s actually essential, hence the P1 priority. Standard threads available are a must as well so they won’t go away.

Perfect !

luzpaz, 2025-12-19:
@alfrix any chance you can weigh-in on this ?

maxwxyz, 2026-11-07:
Part of this is implemented in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/a199895b64276224113b2823c17b91bfbf110894

The Hole panel shows the pitch for the selected standard thread. It's read-only, custom pitch values are still not supported.