FreeCAD/FreeCAD #15732: DXF IMPORT FILE LOCKS UP USING WINDOWS 0.22Dev_38375. IMPORTS OK WITH LATEST 0.21 STABLE RELEASE

greggs123, 2024-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Simple drill map .dxf exported from Kicad used.
Tried several times with FreeCad dev version. Same result.
Imports OK with latest FreeCad 0.21 stable release. Also imports OK using TurboCad.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.14
  * Help 1.0.3
  * sheetmetal 0.4.20
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-02:
Please re-write the title of the ticket (remove the ALLCAPS) and zip the .FCstd file and drag it into the github text box to attach to this ticket for us to reproduce

greggs123, 2024-08-03:
There is no .FCstd file.  What I'm doing is attempting to IMPORT a .dxf file into a NEW part file. That is, File > New. I then go to File > Import  and select the dxf file. FreeCad will then lock up.

I was going to attach the dxf file in question, but this dxf file type is not supported by your GIT tool. dxf now zipped and attached.
[Learn Kicad 8 in 45 minutes-drl_map.zip](https://github.com/user-attachments/files/16476877/Learn.Kicad.8.in.45.minutes-drl_map.zip)

Here are some of my PC specs:
Device name	xxxxxxxxxxxxxxx
Processor	Intel(R) Core(TM) i7-6820HQ CPU @ 2.70GHz   2.71 GHz
Installed RAM	32.0 GB (31.9 GB usable)
Device ID	B1615EE4-BC1F-41ED-888F-4BDC2491AF03
Product ID	00342-50430-17220-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
Edition	Windows 10 Pro
Version	22H2
Installed on	‎3/‎21/‎2021
OS build	19045.4651
Experience	Windows Feature Experience Pack 1000.19060.1000.0

yorikvanhavre, 2024-08-05:
The file fails to open for me (hangs indefinitely), but still not sure what the problem is...

kpemartin, 2024-09-15:
This appears to be a performance issue. The drawing contains about 450 lines and circles representing the PCB itself, and about 1700 lines to build the text at the bottom identifying the holes sizes and counts.
It imports in about 4 or 5 seconds (just counting in my head) if "Use layers" is off, but about 2 minutes if "Use layers" is on. This implies it takes 120 seconds to move 200 shapes to a drawing layer, about 600ms each! This is assuming it is the same amount of time for each layer assignment. If it somehow depends on how many objects are already in the layer, this is a different matter.

Indeed, if I use a text editor to chop out about half the entities in the DXF file, it takes 24 seconds to import, about one quarter of the time, implying there is some sort of n-squared time usage here.

kpemartin, 2024-09-15:
Custom instrumentation code wrapped around the `Layer.addObject` Python method call reveals that the processor time (as reported by `<time.h>`'s `clock` function) required varies from 1-2ms for the first call gradually (and apparently linearly) increasing to 40ms for the last one (with the file shortened as mentioned above).
This doesn't really amount to the total real time observed but that might be an artifact of having a multi-processor system confounding real time vs processor time.
However it clearly indicates that there will be an undesirable n-squared effect here.

kpemartin, 2024-09-15:
The container the layer uses for the objects in the layer is stored in its Group property, and appears to be a App::PropertyLinkListHidden, and this even mentioning lists sets off alarms in my head since `Layer.addObject` starts with the Python code `if child not in Group` which would be an order-N operation.

It looks like each time this `addObject` is called, the C++ code in `App::PropertyLinkList::getPyObject()` creates a fresh `Py::List` object, filling it with all the current layer contents (an order-N operation), which is saved in a local of `Layer::addObject`, then iterated by the Python `in` operator (an order-N operation).
Then, once it has assured itself the object is not already in the layer, it adds the object to the group, and assigns this back to the layer's Group property (not sure what this amounts to in terms of cost).

There is a comment where the `Py::List` is created that says the code should really create a Tuple instead, which perhaps could be faster, but this still seems like a clunky way of doing things.

Poking around in the debugger implies `getPyObject` is called 6 times as part of `Layer.addObject` and each time there is one more iteration of the inner loop as the layer accumulates more contents.

greggs123, 2024-09-15:
Well, it sounds like you’re on the hot trail of what I would call
“abstraction hell”.
When it comes to software, I’m mostly in the Embedded C space. And I’m not
real fond of “abstractions”. But desktop is obviously different.
I’m assuming that the import code drastically changed from the current or
last stable release?
Anyway thanks for the details and good luck!

kpemartin, 2024-09-17:
I've found the real problem: Each time an object is added to a Layer, the Layer's Group property is treated as if it had changed entirely (even though the change is really just a single added element), and so this Group is iterated over a total of 6 times, once (in Layer.addObject) so see if the object is already in the layer, and five times (driven by ViewProviderLayer.updateData, calling ViewProviderLayer.onChanged, calling ViewProviderLayer.change_view_properties) to update properties from the layer to the contained objects.

As a result, adding children one at a time is an O(n^2) operation, taking minutes when there are a thousand or so children.

It looks like the legacy importer would have the same problem since it also calls Layer.addObject one object at a time.

yorikvanhavre, 2024-09-17:
Amazing deduction work @kpemartin!

kpemartin, 2024-09-17:
The legacy importer indeed has the same problem. It imports the original example in 850 seconds, and the truncated example in 136 seconds. The truncated example is a bit less than half the original (I just estimated halfway), and a bit of math reveals exactly the same n-squared behaviour.

I don't intend to fix the legacy importer so I'll open another issue for it.

commit c9887d94 (2024-09-23): DXF: Place objects in layer all at once rather than one at a time to improve DXF import speed dramatically. (#16596)

* Place objects in layer all at once rather than one at a time.
This reduces (by a factor of the number of objects in the layer) the number of times that the layer contents are traversed to set the properties of the contained objects.
This means the layer insertion of O(n) rather then O(n^2) on the number of objects.
Also remove a loop invariant in view_layer so the chidren are not traversed if colours or appearnces are not inherited from the layer.
Fixes #15732

* [pre-commit. [... cut]
commit 238c68d8 (2024-09-23): DXF: Place objects in layer all at once rather than one at a time to improve DXF import speed dramatically. (#16596)

* Place objects in layer all at once rather than one at a time.
This reduces (by a factor of the number of objects in the layer) the number of times that the layer contents are traversed to set the properties of the contained objects.
This means the layer insertion of O(n) rather then O(n^2) on the number of objects.
Also remove a loop invariant in view_layer so the chidren are not traversed if colours or appearnces are not inherited from the layer.
Fixes #15732

* [pre-commit. [... cut]
closed 2024-09-23 by yorikvanhavre