FreeCAD/FreeCAD #15735: Point in sketch as loft profile won't work in dev, but works in stable.

drew585, 2024-08-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Image from Dev build, pre-problemt:
[image]

But when adding a point on a sketch as another profile:
If point is construction: "Loft: invalid section link"
If point is geometry: "Loft: profile shape is not a vertex, edge, wire nor face"

I tried opening the file in FC 0.21.2 and it works after clicking refresh.
[image]

I then opened in Dev again and before clicking anything see this:
[image]

But if I click refresh to get rid of the check marks, it goes back to error state.

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.25
```

### Subproject(s) affected?

Part

### Anything else?

[example11.2FCStd.FCStd.zip](https://github.com/user-attachments/files/16475555/example11.2FCStd.FCStd.zip)

I put the freecad file in a zip since github does not allow .fcstd files for upload.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-08-02:
Good to see a Mint 22 user, confirmed bug, I resorted to what we used to do back in 0.17 days and use a _Draft>Point_ as a workaround, just rename the file from zip to FCStd.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
```

[example11_Syres.zip](https://github.com/user-attachments/files/16475799/example11_Syres.zip)

CalligaroV, 2024-08-17:
Hi @drew585 and @Syres916

I opened a PR that should solve this issue (PR#15913).
In my tests this looks woriking.

Can you please try it and give me a feedback?

TIA!

Syres916, 2024-08-17:
Thanks for the efforts @CalligaroV , this code change appears to affect a lot of operations from multiple workbenches, just for an example if I was applying a _Draft>Path Array_ to a chain link following the Sketch wire of the chain path around a set of sprockets where there isn't a single point involved.
By all means correct me if my assumption is wrong but if it is correct, I'll need some time to check a variety of files aren't negatively impacted.

CalligaroV, 2024-08-17:
Thanks for the feedback @Syres916!

Do you have a file to share that will break because of my proposed modification?
It will help me to figure out what's going on.

TIA!

drew585, 2024-08-18:
@CalligaroV Sorry i'm not familiar on how to use a pull request. I'm not a developer. Just a long time professional Solidworks user teaching myself FreeCAD for personal use and reporting bugs along the way.

CalligaroV, 2024-08-18:
@drew585, don't worry.

If I understood correctly @Syres916 already tried my proposed fix and found issues with other operations/models.
I'll wait for more files to test against my modification and see if/where we'll need other fixes.

Anyway, thanks for reporting this issue!

Syres916, 2024-08-18:
@CalligaroV while I've had a absolute nightmare today with some files, literally hours rebuilding sketches and surfaces due to new more restrictive practices, none of the errors are due to your PR therefore I've no issue with it being set for merge, thanks again.

Something tells me I won't be getting rid of my 0.21.x build for a few years, I just can't justify that amount of time working on customers' files when they only want small changes making to a model.

CalligaroV, 2024-08-19:
Thanks @Syres916 for the feedback!

I'll put #15913 out of draft then.

It's a pity you're facing so many issues with "old" models, probably it will make sense to open new issues regarding backwards compatibility.
ATM however I think those will be analyzed in a post 1.0 dev cycle.

Thanks again!

PR #15913 (2024-09-02): Part/Toponaming: Import TopoShape::prepareProfiles() from LS3
fixes #15735