FreeCAD/FreeCAD #15738: Deleting external geometry messes up the sketch

nazar-pc, 2024-08-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After https://github.com/FreeCAD/FreeCAD/issues/15662 was resolved by https://github.com/FreeCAD/FreeCAD/pull/15712 I think I have another weird case.

I have a fully constrained sketch, trying to delete external geometry that has no associated constraints (`ExternalEdge4` in `MotherboardTrayBase` sketch in `MotherboardTray.FCStd` in attached archive) and it results in things shifting and more unconstrained stuff appearing all of a sudden, which I suspect is a toponaming problem.

Here is a demo:

https://github.com/user-attachments/assets/ce518d0a-0fb8-4044-b459-9945a20d607a

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git) AppImage
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-03:
I think #15712 is still not in the weekly 38385, at least I cannot delete external geometry. So please test this issue next week.

luzpaz, 2024-08-03:
@maxwxyz  an aside, since we are close to ReleaseCandidate releases maybe it's worth everyone's time to either generate AppImages/Windows/macOS dev builds more frequently or even manually?

nazar-pc, 2024-08-03:
> I think https://github.com/FreeCAD/FreeCAD/pull/15712 is still not in the weekly 38385, at least I cannot delete external geometry. So please test this issue next week.

I did check the commit hash and it was corresponding to `main` as of the commit where https://github.com/FreeCAD/FreeCAD/pull/15712 was merged (someone triggered it ~4 hours after merging that PR), so it should be there unless https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds has the wrong information in `freecad_version.txt`.

FEA-eng, 2024-08-03:
I just compare the weekly build number (**38385** in this case) with the number shown when selecting a commit from https://github.com/FreeCAD/FreeCAD/commits/main/ and clicking Browse files. For this commit, it's **38386** so it seems to be very close but not there yet.

[image]

https://forum.freecad.org/viewtopic.php?t=73435#p641091

nazar-pc, 2024-08-03:
> I just compare the weekly build number (38385 in this case) with the number shown when selecting a commit from https://github.com/FreeCAD/FreeCAD/commits/main/ and clicking Browse files. For this commit, it's 38386 so it seems to be very close but not there yet.

That is not correct. Builder is a bit misleading (buggy?), it treats commit index (starting with zero) as commit number, while GitHub shows commit number (starting with 1), hence 1 commit difference. So 38385 is in fact 38386 and you can see in `freecad_version.txt` that it was built from e4b2bf7630f742567e9b2206d591b80f69bbc113 (yes, it is commit 38386), which is exactly the commit where fix for external geometry was merged.

I noticed this number discrepancy yesterday, just didn't report anywhere yet.

maxwxyz, 2024-08-03:
@adrianinsaval FYI

maxwxyz, 2024-08-04:
It's buggy in the last weekly but works when build from main. 
Please test with the next weekly build on Tuesday next week.

bgbsww, 2024-08-04:
Actually, it appears that freecad_version.txt and the .src are updating on each commit, and not at build time.  They are as of the writing now up and 38387, while the build artifacts remain as labeled at 38385. 

This is confusing, but all evidence is that the 38386 version was just missed in the weekly.   It isn't an off by one.

nazar-pc, 2024-08-04:
Got it, thanks folks!

nazar-pc, 2024-08-07:
Work in latest build

closed 2024-08-07 by nazar-pc