FreeCAD/FreeCAD #15746: Stylesheet: Base plane selection could have more contrast

FEA-eng, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It's not a big deal but I think that the selected base plane could have more contrast in Light and Dark themes.

Classic:
[image]

Light:
[image]

Dark:
[image]

Maybe different colors should be used. In Dark mode, there's also a problem with the visibility of the selected plane's name.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-04:
Even worse with simple geometry behind:
[image]

maxwxyz, 2024-10-26:
IMO https://github.com/FreeCAD/FreeCAD/pull/16675 will fix this issue too.

crazyelectron-io, 2024-11-10:
With Freecad 1.0.0RC2 (arm64) on MacOS Sequoia 15.1 using the Light Theme with default settings (except for some font sizes) I can hardly see the difference when selecting a base plane by hovering the mouse over the text label. The blue text and rectangle get ever so slightly lighter blue, but the rectangle is not filled with any visible shade or color.
[image]
I selected the XZ plane; can you tell? :o)

MisterMakerNL, 2024-11-10:
The same selection color is used in sketches which pretty much defined this color. 
It is probably wise to change the default plane color to black in light theme and white in dark theme.
Reasoning is that sketch lines are also white and black in those theme's and making them the same color would solve this conflict.

crazyelectron-io, 2024-11-10:
So the `Enable preselection` color is used here. I'll experiment a bit.
Thnx

MisterMakerNL, 2024-11-10:
The general stance here from the @FreeCAD/design-working-group is we want less color in general. I think everyone would agree that if we could remove the blue from the planes it would be the best action to take.

crazyelectron-io, 2024-11-10:
> The general stance here from the @FreeCAD/design-working-group is we want less color in general.

Fully agree to that! To be honest, I didn't really like the Classic Theme when I started using FreeCAD 0.21 recently.
That's why I start using the Light Theme now that I switched to 1.0.0RC2. Haha, just like the old days when designs where made on paper...

MisterMakerNL, 2024-11-10:
This pr changed them into green red and blue.
https://github.com/FreeCAD/FreeCAD/pull/16675#issuecomment-2466666599

closed 2024-12-06 by chennes