FreeCAD/FreeCAD #15747: Stylesheets: Hidden status of Origin items should be more visible

FEA-eng, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It's a problem with all themes, including the Classic one but it seems to be particularly bad for Dark theme:

[image]

Only the names are more gray but I can barely notice any differences in their icons.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-04:
@MisterMakerNL what say you?

MisterMakerNL, 2024-08-04:
The plane icons seem to be fine but for the axis version it's only a tiny line that changes too white which is not very clear.

maxwxyz, 2025-02-22:
Is this still relevant?