FreeCAD/FreeCAD #15748: Stylesheets: Task panels should have less contrast

FEA-eng, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As an improvement, I would suggest making task panel backgrounds less contrasting. IMO they stand out a bit too much compared with the rest of the Light and Dark theme:

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-04:
@FreeCAD/design-working-group please add your input

Syres916, 2024-08-04:
@FEA-eng if you're not aware, the two places to change in the QSS files if you wanted to adjust your local version and therefore give suggestions, search for:

`QSint--ActionGroup QFrame[class="header"] {`

and

`QSint--ActionGroup QFrame[class="content"] {`

in each is `background-color`

kadet1090, 2024-08-04:
This is basically the same that I pointed out in https://github.com/FreeCAD/FreeCAD/pull/15742. IMHO we should use neutral colors here so in terms of current Light / Dark themes it should be probably some tones of gray. For light I'd use lighter ones than panel background and for dark I'd do darker ones to increase contrast with labels.

maxwxyz, 2025-02-22:
@MisterMakerNL closed with your PR?

MisterMakerNL, 2025-02-22:
@maxwxyz yes! https://github.com/FreeCAD/FreeCAD/pull/19367