FreeCAD/FreeCAD #15751: Stylesheet: Toggled state button uses secondary color

maxwxyz, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Active toggled buttons in the toolbar should use the primary color as if you press them. Currently they use the secondary color (purple) on the default themes. There is also no border between two toggled on buttons, they become one entity (visually).

Here the blue one is the one I just pressed, the purple one are all the other toggled on buttons:
[image]

Here the one with the outline is the hovered one, all purple ones are active:
[image]

@kadet1090 suggestion:
- selected: primary color
- when focussed via keyboard: secondary outline on the button
- when toggled on: just a slightly darker tone or neutral overlay as "pushed in / inset"

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-08-04:
Maybe I'm in a complete minority here but _Disabled_ (if you're going to allocate color) is some form of red:

[image]

kadet1090, 2024-08-04:
The use of red here is absolutely abnormal. Reds should be used in that way to indicate that something is either wrong or dangerous (i.e. deleting stuff, actions that cannot be undone etc) as it is a color that is very focusing and drives a lot of user attention. 

I have not used draft much - is it used to temporarily disable snapping altogether? If so, probably the best mechanism would be to mark this icon as toggled (so "inset" style as mentioned in the description) and make all actions that it disables semi-transparent to clearly indicate that they are not available.

MisterMakerNL, 2024-08-04:
Might indeed be a good idea to make them look like pressed buttons instead of using a color.

Syres916, 2024-08-04:
> Might indeed be a good idea to make them look like pressed buttons instead of using a color.

Seems like the best approach. As someone who does use the Draft snap bar quite a bit the red is there to get my attention that I've left the bar in that state, so yes it's personal and for me it's absolutely normal.

MisterMakerNL, 2024-08-04:
Would this be good:
[image]

MisterMakerNL, 2024-08-04:
or this 
[image]

MisterMakerNL, 2024-08-04:
[image]
and
[image]
in disabled state they stay the same, since the icons are grayed out:
[image]

maxwxyz, 2024-08-04:
maybe even darker for light theme as they are not easily seen with a quick glimpse to the toolbar

maxwxyz, 2024-08-06:
I only see purple outline quickly when unchecking it on mouse down

maxwxyz, 2024-08-06:
@MisterMakerNL I'll reopen this as:
- unchecking a checked button make it appear still active until you move the cursor away
- checking the buttons increases the width of the button. Unchecking them preserved the width but deactivating snapping resets it again

Demo:

https://github.com/user-attachments/assets/32fb54de-c1ee-432c-a1e0-afeac40f7edd

@kadet1090 FYI

MisterMakerNL, 2024-08-06:
Purple is the "pressed" state which you don't really use unless the program feezes up.

closed 2025-07-14 by maxwxyz