FreeCAD/FreeCAD #15755: Simple Additive Loft consumes huge amount of resources. Sometimes have to close the program.

GAZ082, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sometimes i can, sometimes i cant finish this simple loft i'm attaching. Check the video (you may just skip to 2:0 or so).

Remove the .zip in the file.

[Loft Test.FCStd.zip](https://github.com/user-attachments/files/16488134/Loft.Test.FCStd.zip)

https://github.com/user-attachments/assets/d51c0796-1d97-4605-ba7f-e1881e303183

# System Details Report
---

## Report details
- **Date generated:**                              2024-08-04 12:50:06

## Hardware Information:
- **Hardware Model:**                              Gigabyte Technology Co., Ltd. P55-US3L
- **Memory:**                                      16,0 GiB
- **Processor:**                                   Intel® Core™ i5 760  × 4
- **Graphics:**                                    AMD Radeon™ RX 550 / 550 Series
- **Disk Capacity:**                               240,1 GB

## Software Information:
- **Firmware Version:**                            FD
- **OS Name:**                                     Arch Linux
- **OS Build:**                                    rolling
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.10.2-arch1-2

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git) AppImage
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves.backup1713713759.0032098 0.6.29 (Disabled)
  * QuickMeasure 2022.10.28
  * ExtremeProDark 2.9.0
  * Curves.backup1693059780.2922277 0.6.13 (Disabled)
  * CfdOF.backup1710694906.6805208 1.24.9 (Disabled)
  * Assembly4.backup1696800837.713216 0.50.2 (Disabled)
  * Curves.backup1714093860.7275996 0.6.34 (Disabled)
  * OpenTheme 2024.5.16
  * CfdOF.backup1699129791.898633 1.24.7 (Disabled)
  * Assembly4.backup1710902627.637761 0.50.9 (Disabled)
  * Defeaturing.backup1710696629.0288966 1.2.2 (Disabled)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-04:
Could this be related to https://github.com/FreeCAD/FreeCAD/issues/15659 ?
Can you record a macro (start before beginning to reproduce) and end after the issue and attach it here?

GAZ082, 2024-08-04:
@maxwxyz Now i managed to complete the loft (after about 2 minutes).
[Loft Test.FCStd.zip](https://github.com/user-attachments/files/16488632/Loft.Test.FCStd.zip)

Attaching macro and also file with the end result.
[Test2.FCMacro.zip](https://github.com/user-attachments/files/16488630/Test2.FCMacro.zip)

maxwxyz, 2024-08-04:
@bgbsww could this be TNP related. And you have a macro to dive into.

GAZ082, 2024-08-10:
@maxwxyz @bgbsww 

For a simple circle to square loft this seems to work, but the issue arises in my test, which is a circle and a slot.

This is also an issue in 0.21.2

GAZ082, 2024-08-17:
Looks like this is working ok now. It still takes about 20 seconds in my (old) computer, but the ram consumption does skyrokets (about 3gb). I made some crazy lofts as test, all went fine.

[image]

I'll let you guys to decide on the closure of this ticket. Thanks!

@maxwxyz @bgbsww

GAZ082, 2024-08-17:
[image]
Mmmh, perhaps i should open another issue, but this is related i believe. After  clicking ok, the computer just keeps processing, has been like this for the past 2 minutes.

GAZ082, 2024-08-18:
Created a separate issue for the pipe:

https://github.com/FreeCAD/FreeCAD/issues/15951

AIRCAP, 2024-10-18:
fun fact. doing a **Part->Loft** between the 2 sketches is instant. it's just the **PardDesign->Additive loft** that takes forever.

FEA-eng, 2025-07-17:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

GAZ082, 2025-07-17:
@FEA-eng can you attach an example so I can test it as well? Thanks!

luzpaz, 2026-09-06:
So the final verdict is that this is a Core issue not OCC or a combination of the 2 ?

pieterhijma, 2026-09-09:
It's not an OCCT issue.

Lgt2x, 2026-09-09:
needs testing against https://github.com/FreeCAD/FreeCAD/pull/32397