FreeCAD/FreeCAD #15756: Stylesheet: Start page example files not legible with Dark Theme

maxwxyz, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The example file text from the Start page is white on white when using the FreeCAD Dark theme on macOS:
[image]

Using classic, the clicked example file gets a green border:
[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36908 (Git)
Build type: Release
Branch: main
Hash: eb842aadffa2615037c45101ac2575fc79438850
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-08-04:
This is a regression as this was fixed.

luzpaz, 2024-10-05:
This is also apparent on the AppImage

maxwxyz, 2024-10-06:
any progress on this issue?

marcuspollio, 2024-10-07:
I can also reproduce this Stylesheet update issue for the bundled `FreeCAD Light` and `FreeCAD Dark` Themes on a local built with a clean config, leading to the `thumbnailWidget` bad readability on the Dark variant.

The issue is actually only present when the Theme is changed from `isEmpty` or `Classic` to one with a stylesheet (either via the First Start Setup or via the Preferences) until the next FreeCAD launch. At the next launch, the stylesheet is correctly applied for these widgets. But if changing again to `Classic` and back to another one, the issue is present again.

Also note that for Themes other than `Classic`, the styling for the `hovered` and `pressed` states is not applied on my side.

AIRCAP, 2024-10-07:
I just tested this - I can confirm what @marcuspollio wrote - on the recent document previews, the theme is wrong from a theme change until the next freecad restart, 

however - after said next restart I have the theme on the regular buttons wrong:

[image]

```
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 +5 (Git)
Build type: Debug
Branch: restrictiveQuickMeasrure
Hash: 6c84563f02f6d7be7741b86318a6b81a68f49809
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.10
  * CfdOF 1.27.10
  * Curves 0.6.48
  * AirPlaneDesign 0.4.1

```

AIRCAP, 2024-10-07:
additional info: when I am on "no style sheet" - then on hovering over the recent files or examples gives the hovered over file a blue outline

when i switch to dark mode - there is no such outline on either "recent" nor on "examples" - no matter whether the button is white or black

however the blue hover-outline does work on the always white new file buttons

MisterMakerNL, 2024-10-07:
The text color is probably caused because at first boot https://github.com/FreeCAD/FreeCAD/blob/32e09d9554db26414697ed19e0189ccab4ea9029/src/Gui/PreferencePacks/FreeCAD%20Classic/FreeCAD%20Classic.cfg#L83 isn't set to 0.

MisterMakerNL, 2024-10-07:
I btw have no idea why the hoover isn't working @wwmayer could you have a look, since you did most of that work?

adrianinsaval, 2024-10-07:
https://github.com/FreeCAD/FreeCAD/blob/13d00f98841f2e4e038a765ad4897781a30270e4/src/Mod/Start/Gui/StartView.cpp#L232
this should default to false I believe

marcuspollio, 2024-10-07:
> https://github.com/FreeCAD/FreeCAD/blob/13d00f98841f2e4e038a765ad4897781a30270e4/src/Mod/Start/Gui/StartView.cpp#L232
> 
> this should default to false I believe

Hi there, I tried @adrianinsaval suggested fix here, and it does indeed solve some of the issues for a clean start (clean config) when changing to one of the bundled `FreeCAD Light` or `FreeCAD Dark` Themes. But as soon as you choose the `Classic` theme again, the issue is back.

Also,  the `hovered` and `pressed` states for these two non-Classic themes do nothing.

<details><summary>FreeCAD Info</summary>

```
OS: GNU/Linux Ubuntu 22.04 LTS (DE: Gnome/pop)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 +1 (Git)
Build type: Release
Branch: start-style-fix
Hash: 6e6edbf55fd9ea9305ae1627145d9a7300ea8e75
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

```
</details>

luzpaz, 2024-10-14:
bumping for visibility

maxwxyz, 2024-10-19:
@MisterMakerNL I'm unassigning you but you can still submit a PR to fix it. There is a $500 bounty for this issue.

benj5378, 2024-10-19:
I am currently working on this as a bounty

MisterMakerNL, 2024-10-26:
So I am just going to say how I think this dialog should work, instead on how we probably should fix this, since my idea is probably not possible on short time.
So the correct way of making this dialog is to do what most CAD software does. 
Which means it checks a directory for the links/templates that are presented and when pressed on that link, instead of loading the file it loads a copy as a new file.
We also get the options for companies/people to have their own templates. 
We also add the option for addons too add templates.

So instead of having hard-coded buttons we need a simplified Qfiledialog that checks a specific directory, or searchpath and loads the links from their.

benj5378, 2024-10-26:
@MisterMakerNL is this related to the visuals of the start page?

MisterMakerNL, 2024-10-26:
@benj5378 Well a fix is a fix, and I have no clue how hard of all that work is so I am just thinking out loud. Please ignore me if you think it madness doing something like that!

closed 2024-10-28 by yorikvanhavre

chennes, 2024-11-06:
Noting that @benj5378 is eligible to claim the bounty from the FPA. To claim, please send an email to fpa@freecad.org referencing this issue, and your preferred method of payment.

PR #17376 (2024-10-28): [Gui] Make styles update on paint
Fixes #15756. Similar implementation to that already in FileCardDelegate.cpp.
commit 954bae72 (2024-10-28): [Gui] Make styles update on paint (#17376)

* [Gui] Make styles update on paint

Fixes #15756. Similar implementation to that already in FileCardDelegate.cpp

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

---------