FreeCAD/FreeCAD #15757: Stylesheet: Light and Dark indents the hovered entry on dropdown menus

maxwxyz, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using FreeCAD Dark it indents the hovered entry on dropdown menus.
Here is an example from the preferences:
[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36908 (Git)
Build type: Release
Branch: main
Hash: eb842aadffa2615037c45101ac2575fc79438850
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-08-04:
Could be the 2px padding in Qlabel, Could you test the Stylesheet from https://github.com/FreeCAD/FreeCAD/pull/15742

maxwxyz, 2024-09-03:
@MisterMakerNL this did not solve it, still the same issue on the latest weekly
```
OS: macOS 13.6.9
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.8.17
  * CfdOF 1.27.5
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.27
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.24
  * Curves 0.6.45
```

luzpaz, 2024-09-22:
bumping for visibility

MisterMakerNL, 2024-09-22:
This issue is very much a thing in QT 6, and there is also a fix for this in QT6, the thing is if I fix this in stylesheets in QT 5.15 for mac. I'll ruin it is for the other os that do not have issues here.
I also wonder if this bug report was before or after the switch to QTstyle fusion? And so can you test it again @maxwxyz ?
Is this btw a blocker? I don't see this as a mayor issue just an annoyance especially knowing it will get fixed for QT 6+

maxwxyz, 2024-09-23:
The issue was also present in Qt5 (see original post) and fusion style did not change it. IMO not a blocker.

maxwxyz, 2024-09-24:
@kadet1090 @MisterMakerNL I also have this behavior on Windows now, Qt5. Normally I've only had this issue on macOS with Qt5.
[image]

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.22
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5

```

MisterMakerNL, 2024-09-24:
Maybe this is caused by the icon fix since thr check mark. Anf I have too adjust the stylesheets to the new icons.

maxwxyz, 2024-10-05:
any news on this one?

kadet1090, 2025-08-13:
@maxwxyz still relevant?

maxwxyz, 2025-08-14:
No, fixed!

closed 2025-08-14 by maxwxyz