FreeCAD/FreeCAD #15758: Stylesheet: Start page new file buttons have cropped text

maxwxyz, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Start page new file buttons have cropped text, here using FreeCAD light theme:
[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36908 (Git)
Build type: Release
Branch: main
Hash: eb842aadffa2615037c45101ac2575fc79438850
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-08-04:
I noticed it too, but I cannot help you here. Can @Syres916 or @kadet1090 have a look at this?

MisterMakerNL, 2024-08-04:
Not sure if I fix anything but it's gone on my install 
[image]

MisterMakerNL, 2024-08-05:
QT 6
[image]
QT 5
[image]

maxwxyz, 2024-08-05:
will test with next weekly

maxwxyz, 2024-08-06:
Works on windows:
[image]
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23

```

maxwxyz, 2024-08-06:
But still an issue on macOS @MisterMakerNL @kadet1090 
[image]

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

maxwxyz, 2024-08-06:
ah damn the macOS did not update it's weekly @adrianinsaval FYI

adrianinsaval, 2024-08-06:
should be updated now

maxwxyz, 2024-08-06:
it's fixed
[image]

maxwxyz, 2024-08-06:
Reopening, with the Classic theme it is now back on macOS

[image]

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

maxwxyz, 2024-09-03:
Still an issue in the latest weekly 38622
[image]

luzpaz, 2024-10-23:
Is this still happening ?

maxwxyz, 2024-10-23:
yes

PR #19901 (2025-03-06): Start: fix text visbility
fixes #15758

fixes #16843

due to the removal of render() and the refactor probably fixes this two
#17771
#19749

## Freecad Classic when system is dark:
[image]
commit ce2cae5b (2025-03-06): start: fix new file button layout size

fixes #15758
closed 2025-03-06 by chennes